ct-Open-Source/tuya-convert #576: Cannot connect to IP camera (SC002-WA2)

thi-baut, 2020-03-27:
Hi, 

I'm trying to connect to my IP camera to dump the firmware (and further customise it). I reset the device and put it in EZ mode (blinks fast). Tuya-convert seems to interact with it as it emits a noise like if it was being paired but then nothing happens after the smart config (see logs below)

Model: SC002-WA2 (REV 3) - based on what I read on the board after opening the camera

If it helps, this is what I got as a result of an nmap scan:
- OS CPE: cpe:/o:linux:linux_kernel:2.4.37 cpe:/o:linux:linux_kernel:3.2
- OS details: DD-WRT v24-sp2 (Linux 2.4.37), Linux 3.2
- Only port 6668 seems open

> Starting smart config pairing procedure
> Waiting for the device to install the intermediate firmware
> Put device in EZ config mode (blinking fast)
> Sending SSID                  vtrust-flash
> Sending wifiPassword          
> Sending token                 00000000
> Sending secret                0101
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> ..................
> SmartConfig complete.
> Resending SmartConfig Packets

Any idea on how to progress? I successfully flashed other compatible devices before (mostly light bulbs)

kueblc, 2020-03-27:
Sorry, `tuya-convert` only supports ESP8266

closed 2020-03-27 by kueblc

rkkoszewski, 2020-04-16:
> Hi,
>
> I'm trying to connect to my IP camera to dump the firmware (and further customise it). I reset the device and put it in EZ mode (blinks fast). Tuya-convert seems to interact with it as it emits [...]

Did you finally manage to get any further? I'm trying to do the same.. Uboot in the device requires password, I'm hoping to get more info from the firmware..

thi-baut, 2020-04-16:
Password could be "12345678" based on this topic: https://github.com/AMoo-Miki/homebridge-tuya-lan/issues/4 - we are a small group trying to get an RTSP stream different camera models. I'll read more about uboot, I don't know how to use it. Also started a thread here to get ideas: https://www.reddit.com/r/homeautomation/comments/ftjrrk/tuya_sc002wa2_integration/ as this is not really related to tuya-convert

> Did you finally manage to get any further? I'm trying to do the same.. Uboot in the device requires password, I'm hoping to get more info from the firmware..

cjj25, 2022-05-20:
I've just released an RTSP server patch that's applied via SD card here: [cjj25/RTS3903N-Tuya-RTSPServer](https://github.com/cjj25/RTS3903N-Tuya-RTSPServer)