ct-Open-Source/tuya-convert #578: Error when flashing RGB Smart Plug

mhofer2011, 2020-03-29:
Hello,

I tried flashing a Chinease Tuya Smart Socket - exactly similar with this one:

[https://github.com/arendst/Tasmota/wiki/RGB-Smart-Plug-16A]

but with the the current version 8.1.0

However, after the first flash step the smart socket isn't responding anymore. I haven't found any way for a fast reset yet. 

So my first question is: Does anyone know a way to reset the device without disasembling?
If not, how do I reset the device after deasembling?
I will provide the log files. Can anyone help me to understand it?

I read that Tuya is changing its firmware to prevent flashing Tasmota. Will this result in a brick oder just stop the installation.

Kind regards,

Mario

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4398872/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4398874/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4398875/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4398876/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4398877/smarthack-web.log)

kueblc, 2020-03-29:
Your device is running the intermediate firmware. Restart the process, plug the device in, and press enter. Do not connect a phone or attempt to put the device in pairing mode.

mhofer2011, 2020-03-30:
Thanx - it work!

closed 2020-03-30 by mhofer2011