ct-Open-Source/tuya-convert #579: Stuck on flashing minimal build on RGB plugn (AP only with ESP-xxxxx)

mhofer2011, 2020-03-30:
Hello,

I just solved one problem - and out of my own stupidity - I flashed with a second device not the tasmota basic bin but the minimal bin I downloaded for local update language version before.
I was a bit confused and didn't check twice: the script does not recognise bin files with several dots like tasmota.8.2.0.bin. So it didn’t offer all files within the folder.
Now I have the problem as other before (1 year ago), Flashing was successful but the device opens only a AP with ESP-XXXXXX. With gateway 192.168.4.1 and my PC getting 192.168.4.2 if connected. However, no UI to update / access. No ssh.

Restarting flashing fails - of course as the ESP is individually generated.

Is there a way to start the process anyway? Probably by changing the name of the connecting Wifi SSID to "ESP-XXXX" in one of the scripts. As I am not a pro on this, any help would be highly appreciated. Bad luck number 2: the plug is not easily to open...

Regards, 

Mario

kueblc, 2020-03-30:
`tuya-convert` only works on stock Tuya firmware. It cannot install firmware on a device once it is running a third party firmware like Tasmota. You will need to seek help from the Tasmota community to figure out how to recover from a minimal installation on a blank device.

> the script does not recognise bin files with several dots like tasmota.8.2.0.bin. So it didn’t offer all files within the folder.

This however sounds like a bug in `tuya-convert` that I can address, thank you for bringing this to my attention. I will push a fix out shortly.

closed 2020-03-30 by kueblc

solanek, 2022-10-04:
@mhofer2011 did you manage to solve the problem?

mhofer2011, 2022-10-04:
No I did not - but any help would still be highly appreciated.

Regards, Mario