ct-Open-Source/tuya-convert #580: Loratap SC500WSC-V2: Device did not appear with the intermediate firmware

kaybau, 2020-04-05:
Hello,
I tried several times to flash SC500WSC-V2 OTA with tyua-convert.
tuya-convert itself is working fine on my Raspi, I flashed succesfully other devices.
But for SC500WSC-V2 I got the message that the device did not appear. The device was blinking quickly in pairing mode and changed the blinking frequence once the flash program connected via VTRUST.
To check if the correct chip is inside the device I checked the device agains https://templates.blakadder.com/loratap_SC500W.html . It has the right chip onboard.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4433370/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4433371/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4433372/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4433373/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4433374/smarthack-wifi.log)
[image]

The connection of the device is shown in wifi log.
Any idea if flashing OTA is possible with this device?
Cheerrs,
Kay

kueblc, 2020-04-05:
Unfortunately it looks like you have a patched firmware that is not compatible.

Follow #483 for more information and hopefully an eventual resolution.

closed 2020-04-05 by kueblc

kaybau, 2020-04-07:
Will monitor 483 - if there is any solution