ct-Open-Source/tuya-convert #581: I can't find my device after flashing what now?

Jelmer-Meurisse, 2020-04-08:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2020-04-08:
Please elaborate, what do you mean you can't find it?

kinselp, 2020-04-11:
I have had a similar experience twice now. Tuya convert works fine, update ssid/pew in tasmota thru 192.168.4.1 and then the device doesn’t show up.

For some reason the device doesn’t seem to go to dhcp and latches on to an odd ip range outside my router range (10.##.##.##.##) Found the device connecting correctly to AP thru unify APP app, then was able to get MAC address and thru the router force it to the correct ip range thru dhcp reservation.  

Hope I explained that well enough.

Toki99, 2020-04-27:
I think I have the Problem with da Blitzwolf SPH6 Device

Toki99, 2020-04-29:
Hi, @kinselp Do use the Modul with the external antenna?
Check that the Module not as a Acces Point in your Network.

kueblc, 2020-06-20:
Closing until we receive more information

closed 2020-06-20 by kueblc