ct-Open-Source/tuya-convert #582: Gosund SP1 switches off as soon as flashing begings?

membersound, 2020-04-08:
I almost succeeded in flashing my first Gosund SP1 device. But I'm stuck as follows: as soon as tuya converter starts to execute and reaches the first `SmartConfig complete.` output, the SP1 exits the pairing mode (you can hear it, and the led on top changes from flashing-blue to constant-red).

All future retried fail of course then. How can I find the rootcause why this happens?

I'm running ubuntu 18.04, can see the active hotspot and I'm connected with an android phone all the time.

```
docker-compose exec tuya start
tuya-convert v2.4.3
Checking for network interface wlp58s0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
sudo: ufw: command not found
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
........
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
No screen session found.
Closing AP
Exiting...

```

Docker container logs:
```
Apr  8 13:35:18 vtrust-flash dnsmasq[2387]: started, version 2.79 cachesize 150
Apr  8 13:35:18 vtrust-flash dnsmasq[2387]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
Apr  8 13:35:18 vtrust-flash dnsmasq-dhcp[2387]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Apr  8 13:35:18 vtrust-flash dnsmasq-dhcp[2387]: DHCP, sockets bound exclusively to interface wlp58s0
Apr  8 13:35:18 vtrust-flash dnsmasq[2387]: read /etc/hosts - 8 addresses
Apr  8 13:35:22 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: authenticated
Apr  8 13:35:22 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: associated (aid 1)
Apr  8 13:35:22 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca RADIUS: starting accounting session 1C3E589A4E4D58B7
Apr  8 13:35:22 vtrust-flash dnsmasq-dhcp[2387]: DHCPDISCOVER(wlp58s0) 60:01:94:f4:40:ca 
Apr  8 13:35:22 vtrust-flash dnsmasq-dhcp[2387]: DHCPOFFER(wlp58s0) 10.42.42.13 60:01:94:f4:40:ca 
Apr  8 13:35:22 vtrust-flash dnsmasq-dhcp[2387]: DHCPREQUEST(wlp58s0) 10.42.42.13 60:01:94:f4:40:ca 
Apr  8 13:35:22 vtrust-flash dnsmasq-dhcp[2387]: DHCPACK(wlp58s0) 10.42.42.13 60:01:94:f4:40:ca ESP_F440CA
Apr  8 13:35:35 vtrust-flash hostapd: wlp58s0: STA 14:5f:94:dc:dc:d3 IEEE 802.11: authenticated
Apr  8 13:35:35 vtrust-flash hostapd: wlp58s0: STA 14:5f:94:dc:dc:d3 IEEE 802.11: associated (aid 2)
Apr  8 13:35:35 vtrust-flash hostapd: wlp58s0: STA 14:5f:94:dc:dc:d3 RADIUS: starting accounting session 9C832FD9D1095BC3
Apr  8 13:35:35 vtrust-flash dnsmasq-dhcp[2387]: DHCPDISCOVER(wlp58s0) 14:5f:94:dc:dc:d3 
Apr  8 13:35:35 vtrust-flash dnsmasq-dhcp[2387]: DHCPOFFER(wlp58s0) 10.42.42.38 14:5f:94:dc:dc:d3 
Apr  8 13:35:35 vtrust-flash dnsmasq-dhcp[2387]: DHCPREQUEST(wlp58s0) 10.42.42.38 14:5f:94:dc:dc:d3 
Apr  8 13:35:35 vtrust-flash dnsmasq-dhcp[2387]: DHCPACK(wlp58s0) 10.42.42.38 14:5f:94:dc:dc:d3 HUAWEI_P10-d45740040ba7f9
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:03 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: disassociated
Apr  8 13:36:04 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Apr  8 13:36:21 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: authenticated
Apr  8 13:36:21 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca IEEE 802.11: associated (aid 1)
Apr  8 13:36:21 vtrust-flash hostapd: wlp58s0: STA 60:01:94:f4:40:ca RADIUS: starting accounting session D97EF8E86B40833C
Apr  8 13:36:21 vtrust-flash dnsmasq-dhcp[2387]: DHCPDISCOVER(wlp58s0) 60:01:94:f4:40:ca 
Apr  8 13:36:21 vtrust-flash dnsmasq-dhcp[2387]: DHCPOFFER(wlp58s0) 10.42.42.13 60:01:94:f4:40:ca 
Apr  8 13:36:21 vtrust-flash dnsmasq-dhcp[2387]: DHCPREQUEST(wlp58s0) 10.42.42.13 60:01:94:f4:40:ca 
Apr  8 13:36:21 vtrust-flash dnsmasq-dhcp[2387]: DHCPACK(wlp58s0) 10.42.42.13 60:01:94:f4:40:ca ESP_F440CA
```

kueblc, 2020-04-08:
Please share your log files

membersound, 2020-04-08:
Where can I find the logs? I assumed in /tuya-converter/scripts. But there are no logfiles. Do I have to extract them from the docker container somehow?

kueblc, 2020-04-08:
It's also possible they are in the `tuya-convert/backups` directory

membersound, 2020-04-08:
There are no logs in the execution directly?

```
/tmp/tuya-convert$ find | sed 's|[^/]*/|- |g'
.
- scripts
- - psk-frontend.py
- - setup_checks.sh
- - mq_pub_15.py
- - firmware_picker.sh
- - tuya-discovery.py
- - setup_ap.sh
- - hostapd.conf
- - smartconfig
- - - crc.py
- - - smartconfig.py
- - - main.py
- - - broadcast.py
- - - multicast.py
- - fake-registration-server.py
- install_prereq.sh
- docker-compose.yml
- start_flash.sh
- docker
- - bin
- - - stop
- - - start
- - - config.sh
- - docker-compose.sample.yml
- .github
- - FUNDING.yml
- - ISSUE_TEMPLATE.md
- .gitignore
- LICENSE
- .git
- - info
- - - exclude
- - index
- - description
- - branches
- - hooks
- - - pre-applypatch.sample
- - - pre-rebase.sample
- - - pre-commit.sample
- - - update.sample
- - - prepare-commit-msg.sample
- - - pre-receive.sample
- - - pre-push.sample
- - - applypatch-msg.sample
- - - post-update.sample
- - - commit-msg.sample
- - - fsmonitor-watchman.sample
- - logs
- - - refs
- - - - heads
- - - - - master
- - - - remotes
- - - - - origin
- - - - - - HEAD
- - - HEAD
- - config
- - packed-refs
- - objects
- - - info
- - - pack
- - - - pack-d0602c9dfda7ab0f300252123450b11b94bc8fc4.pack
- - - - pack-d0602c9dfda7ab0f300252123450b11b94bc8fc4.idx
- - refs
- - - heads
- - - - master
- - - tags
- - - remotes
- - - - origin
- - - - - HEAD
- - HEAD
- README.md
- Dockerfile
- files
- - espurna.bin
- - upgrade.bin
- - tasmota.bin
- data
- - backups
- config.txt
```

kueblc, 2020-04-08:
@Cyber1000 do you know why logs would not be showing up in docker?

membersound, 2020-04-08:
Or do I have login into the docker container and fetch the logs from there? I mean, how should docker know where to write logs if no log directory is shared in `docker-compose`?

The only thing I have is:
```
...
 volumes:
      - ./data/backups:/tmp/tuya-convert/data/backups
```

kueblc, 2020-04-08:
@membersound regarding

> the SP1 exits the pairing mode (you can hear it, and the led on top changes from flashing-blue to constant-red)

This is expected, it should happen. Now that it is paired, you don't need to connect a phone or put the device back in pairing mode.

> All future retried fail of course then

What do you mean "of course"? How did you retry? Don't attempt to put the device back in pairing mode, if that is what you mean.

Also check that the device is not already running the intermediate firmware. You might see a `vtrust-recovery` AP when the device is plugged in. Another way you can tell is if the device does not respond to button presses.

kueblc, 2020-04-08:
> Or do I have login into the docker container and fetch the logs from there? I mean, how should docker know where to write logs if no log directory is shared in `docker-compose`?

I don't use docker and cannot provide support for it, @Cyber1000 wrote this addition and will comment if they have anything to add.

membersound, 2020-04-08:
> This is expected, it should happen. Now that it is paired, you don't need to connect a phone or put the device back in pairing mode.

Ok I thought this would mean a "disconnect" and thus any further retries failed.

Then I'd be really interested in finding the logs. I don't see any vtrust-recovery AP, so I assume the devices are still plain stock.

membersound, 2020-04-08:
I extracted the logs as follows:
Log into the docker container:
`docker exec -it tuyaconvert_tuya_1 /bin/bash`

Copy logs to shared volume (in my case):
`cp /usr/bin/tuya-convert/scripts/smarthack* /tmp/tuya-convert/data/backups/`

With this, in the -web.logs I could find that port 80 was occupied by my local `apache2`. I did not know it has to be free. And unfortunately tuya-convert does not report this directly to the CLI.

membersound, 2020-04-08:
I'm leaving this issue open for now, as you maybe want to have a deeper look on why logs are not shared by docker. Or maybe this is not intended yet and could be a feature?

membersound, 2020-04-08:
I got the solution: I switched the shared volume definition. I thought the later path would be the path on my host. But in fact it's the docker path, which I had overridden with `- ./data/backups:/tmp/tuya-convert/data/backups` (which is totally wrong). Thus logs missing.

closed 2020-04-08 by membersound

kueblc, 2020-04-08:
> With this, in the -web.logs I could find that port 80 was occupied by my local `apache2`. I did not know it has to be free. And unfortunately tuya-convert does not report this directly to the CLI.

If run directly on the host, `tuya-convert` will detect this and let the user decide how to proceed. I am not sure this is possible inside a VM, someone experienced with docker will have to answer this.

> I'm leaving this issue open for now, as you maybe want to have a deeper look on why logs are not shared by docker. Or maybe this is not intended yet and could be a feature?

@Cyber1000 could you look into this?

> Though I'm glad the flashing worked!

Very happy to hear it worked out, thank you for sharing your experience :+1:

Cyber1000, 2020-04-11:
Sorry for the late reply.
I executed the docker-container the last time in December 2019 and I have all log files in the backup-Folder:

At least I have 4 log-Files in it:
[image]

I don't knoq exactly if there were bigger changes, but I've looked in https://github.com/ct-Open-Source/tuya-convert/blob/master/start_flash.sh (line 123) and the logs are still moved into the backupfolder (which is shared by docker)

Cyber1000, 2020-04-11:
Ah I see the last post of @membersound says:

> I thought the later path would be the path on my host. But in fact it's the docker path, which I had overridden with - ./data/backups:/tmp/tuya-convert/data/backups (which is totally wrong). Thus logs missing.

From the sample file https://github.com/ct-Open-Source/tuya-convert/blob/master/docker/docker-compose.sample.yml:
`- ./data/backups:/usr/bin/tuya-convert/backups`

- You should never change the path after the :, cause that's the path within docker (which doesn't change), if you change this you won't get any files you might expect.
- You may change the path before the :, cause this is the path on your local computer.
- In this case it's a relative path ./data/backups, relative to the directory where docker-compose.yml exists, but it may also be an absolute path or a docker-volume. 

So the problem here was the wrong usage of the volumes section, I'll think about it, maybe some comments to https://github.com/ct-Open-Source/tuya-convert/blob/master/docker/docker-compose.sample.yml would be the most effective to prevent this in future.