ct-Open-Source/tuya-convert #583: ModuleNotFoundError: No module named 'Cryptodome'

ndfred, 2020-04-10:
When flashing a new device with

```
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/fred/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/fred/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
```

This is with the latest checkout (release v2.4.3) on my Raspberry Pi 4. I'll try and fix the issue and report back.

ndfred, 2020-04-10:
It looks like `install_prereq.sh` is installing `pycryptodomex` but not `pycryptodome`, I'll try and install that and re-run the process:

`$ sudo -H python3 -m pip install --upgrade pycryptodome`

ndfred, 2020-04-10:
I was actually in a python virtual env which is why python wasn't seeing the packages 🤦‍♂️

It might be best to install the tuya-convert python packages in a virtual env to make it cleaner, and that may avoid issues like this in the future. I'll close this as this was really my mistake.

closed 2020-04-10 by ndfred

kueblc, 2020-04-10:
> It looks like `install_prereq.sh` is installing `pycryptodomex` but not `pycryptodome`

`pycryptodomex` is intentional. `pycryptodomex` and `pycryptodome` have an important distinction, while `pycryptodome` strives to be a drop in replacement for the old `Crypto` package, `pycryptodomex` uses the `Cryptodome` name. This ensures that if you have the deprecated `Crypto` package installed, we use the correct package.

Might your issues be at all related to #575? Or rather that the issue presented in #575 is related to the python virtual env?

> It might be best to install the tuya-convert python packages in a virtual env to make it cleaner, and that may avoid issues like this in the future. I'll close this as this was really my mistake.

Feel free to submit a PR :)

ministryofsillywalks, 2020-05-15:
I'm having the same issue however I didn't use any virtual environment to install tuya-convert.
Sadly I have no clue what the hell is going on with python and whatnot ;)
Any nudge in the right direction would be great.

micooke, 2020-06-23:
I had this same issue, which was a result of installing the prerequisites as sudo - and then running start_flash.sh without sudo.

So to reword this - manually installing the prerequisites from install_prereq.sh without sudo fixed my issue (or running start_flash.sh with sudo).

RolandTalstra, 2020-07-29:
I'm experiencing the same problem, however sudo ./start_flash.sh doesn't work for me, the error remains the same

I ran the prereq's again, and re-installed tuya_convert, but to no avail.

Any more tips?

abdalians, 2020-08-02:
getting the same error, any idea how to fix this?

JohannCR, 2021-01-17:
Same error. Any news ?

KDMcMullan, 2021-02-19:
Oooh, this is odd. I've not had this error before, but I have it now. I haven't used tuya-convert in a few months and I can't imagine what I might've done in the meantime!

KDMcMullan, 2021-02-19:
I've just done a complete fresh install on a "virgin" Raspberry Pi and got the same error message. Are there any clues what might be going on? There are still a remarkable number of devices (and users) around which can really benefit from this method, if it's still supported. Thanks.

TungstenE2, 2021-04-09:
same here, using Pi 3+ running Buster

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/pi/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
........................................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
No ESP82xx based devices connected according to your wifi log.
Here is a list of all the MAC addresses that connected:
0c:8f:ff:4xxxxxxxxxxx

If you see your IoT device in this list, it is not an ESP82xx based device.
Otherwise, another issue may be preventing it from connecting.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] N
======================================================
Cleaning up...
No screen session found.
No screen session found.
No screen session found.
Closing AP
Exiting...
pi@raspberrypi:~/tuya-convert $

TungstenE2, 2021-04-09:
why is this ticket closed with no solution?

also this ticket is closed: https://github.com/fashberg/WThermostatBeca/issues/39

kueblc, 2021-04-09:
> why is this ticket closed with no solution?

@TungstenE2, the original ticket author @ndfred closed this after posting the solution

> I was actually in a python virtual env which is why python wasn't seeing the packages man_facepalming
> 
> It might be best to install the tuya-convert python packages in a virtual env to make it cleaner, and that may avoid issues like this in the future. I'll close this as this was really my mistake.

bachoo786, 2021-06-03:
has anyone tried to resolve this issue? I cant seem to be using tuya convert anymore and its a shame. any help please?

micooke, 2021-06-03:
It's an issue of polluted path. It's and an issue with tuya-convert.
Python is telling you it can't getting this module, which means it is in installed for the python env you are using. Nothing the issue that sudo installs module to a system-wide location, and a non sudo install to a user location.

smizles, 2021-07-05:
I have the exact same behavior when I run `./start_flash.sh` where it errors out with the following message:
```
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/pi/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
```

I originally ran `./install_prereq.sh` and `./start_flash.sh` without `sudo` first, but both scripts asked for `[sudo] password for pi:` at various points in the script. *NOTE:* `./install_prereq.sh` does have this error shown at the end:
```
Collecting git+https://github.com/drbild/sslpsk.git
  Cloning https://github.com/drbild/sslpsk.git to /tmp/pip-req-build-fya70jk6
Please do not run git as root, your regular user account is enough :)

If you need to run git with root rights for some other application than
what comes preinstalled on this image you can remove this sanity check:

    sudo rm /root/bin/git

You might have to restart your login session after doing that.
Command "git clone -q https://github.com/drbild/sslpsk.git /tmp/pip-req-build-y186am_q" failed with error code 1 in None
```

However, I am able to manually run `git clone -q https://github.com/drbild/sslpsk.git /tmp/pip-req-build-y186am_q` after the script finishes without any errors. I also noticed that after I read this thread when I reran it with `sudo` I got basically the same error but the `/tmp/pip-req-build-...` folder changes.

My main roadblock is I get the following error after running `./start_flash.sh`:
```
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/pi/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
........................................................................................................................
Timed out while waiting for the device to (re)connect
```

Per @ndfred 's suggestion above, I tried running `sudo -H python3 -m pip install --upgrade pycryptodome` and (I think) I must not have had it because it executed with the following message:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pycryptodome
  Downloading https://www.piwheels.org/simple/pycryptodome/pycryptodome-3.10.1-cp35-abi3-linux_armv7l.whl (1.8MB)
    100% |████████████████████████████████| 1.8MB 243kB/s
Installing collected packages: pycryptodome
Successfully installed pycryptodome-3.10.1
```

However, again following @ndfred 's comment, I get the following errors after running their respective commands:
```
pi@octopi:~/tuya-convert $ python -m Cryptodome.SelfTest
/usr/bin/python: No module named Cryptodome
pi@octopi:~/tuya-convert $ python3 -m Cryptodome.SelfTest
/usr/bin/python3: Error while finding module specification for 'Cryptodome.SelfTest' (ModuleNotFoundError: No module named 'Cryptodome')
```

I double-checked the version information, and it appears that I have both versions of python:
```
pi@octopi:~/tuya-convert/scripts $ python --version
Python 2.7.16
pi@octopi:~/tuya-convert/scripts $ python3 --version
Python 3.7.3
```

Also, I tried to run `main.py` with manually specifying `python3` and got this error message:
```
pi@octopi:~/tuya-convert/scripts/smartconfig $ python3 main.py
Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/pi/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
```

Here are similar issues I have checked:
1. https://github.com/ct-Open-Source/tuya-convert/issues/643
2. https://github.com/ct-Open-Source/tuya-convert/issues/754
3. https://github.com/ct-Open-Source/tuya-convert/pull/575
4. https://github.com/fashberg/WThermostatBeca/issues/39

Other information:
- I'm running Octopi version 1.6.1
- It's running on a Raspberry Pi 4b 2GB
- `cat /etc/os-release`:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

smizles, 2021-07-05:
UPDATE: I'm getting a different error from `pi@octopi:~/tuya-convert/scripts/smartconfig $ python3 main.py` after running `sudo pip3 install pycryptodomex` and letting that install. I will try to reflash shortly because the error looks like the error is related to the AP not running. :crossed_fingers: 

Sorry for some minor cross-posting, but this other thread was the second most promising: https://github.com/ct-Open-Source/tuya-convert/issues/643

2nd UPDATE: I noticed I still had the following errors in `smarthack-psk.log` and `smarthack-web.log` respectively:
```
Traceback (most recent call last):
  File "./psk-frontend.py", line 6, in <module>
    import sslpsk
ModuleNotFoundError: No module named 'sslpsk'
```
```
Traceback (most recent call last):
  File "./fake-registration-server.py", line 9, in <module>
    import tornado.web
ModuleNotFoundError: No module named 'tornado'
```

Both appear to have been solved by running `sudo pip3 install sslpsk` and `sudo pip3 install tornado` respectively and letting each finish installing before continuing on. I will try to reflash again, but I wanted to update my troubleshooting as I went. BTW, I still get the error `Command "git clone -q https://github.com/drbild/sslpsk.git /tmp/pip-req-build-m7e4a1si" failed with error code 1 in None` when I re-run `./install_prereq.sh`, but at least I'm not getting anymore `ModuleNotFoundError: No module named 'sslpsk'` errors in my `smarthack-psk.log` file

***NOTE:*** This may be an issue with trying to run `tuya-convert` from a spare RPi running Octopi because I have found other threads of people having issues from this distribution - but it should theoretically be supported because Octopi is just a customized distro of Raspberry Pi OS which is itself a Debian distro, right?

micooke, 2021-07-05:
So the install prerequisites isn't quite correct, as there is a flow through of sudo where to the pip commands - so yeah that's not right. But you can get around this by installing them manually. A big hint is that whenever you get a ModuleNotFoundError: No module named ... error, that's directly from python. That module is not on pythons path. Think of installing a python package as sudo as installing it as another user (because it is), so you need to match the install user to whatever user you are running your program as. In general, don't install python packages as sudo unless they are required by a system utility or a python utility that is daemonised.

smizles - you are installing pycryptodome as sudo, sudo -H python3 -m pip install --upgrade pycryptodome then you ran the test as non sudo python3 -m Cryptodome.SelfTest . FYI - generally python is linked to python2.7 (you can test this with python --version ) and python modules are generally python version dependent, but they certainly are when you install them. So use pip3 to install for python3, and pip for python2.7

smizles, 2021-07-09:
Hi @micooke,
Thanks for taking the time to write such a thorough answer! I'll definitely have to try running that last code for manually installing the prerequisite modules as non-sudo, but I think I finally have them because now the script errors out after `Smart config sent. / Resending smart config...`. Maybe this is related to my device's firmware being incompatible (is there anything in any of the log files that might tell me so? I didn't see any glaring anomalies after a quick glance at each).

> @smizles - you are installing pycryptodome as sudo, `sudo -H python3 -m pip install --upgrade pycryptodomex` then you ran the test as non sudo `python3 -m Cryptodome.SelfTest` .

Here's an interesting fact, I actually did ***not*** run either of the two scripts as `sudo` during my first attempt. The system actually paused midway through the script and prompted me for my sudo password (is this a big with the script or some default setting on my Raspberry Pi that I should change?). Anywho, either way the three modules I specified above were never installed (for any user, sudo or not apparently). The modules successfully installed when I manually ran the `pip` command specified in my second post (using sudo in that particular instance was testing a theory somebody else had in another comment).

> FYI - generally python is linked to python2.7 (you can test this with python --version ) and python modules are generally python version dependent, but they certainly are when you install them. So use pip3 to install for python3, and pip for python2.7

Sorry for any miscommunication between my original and update comments. Apparently the modified Raspberry Pi OS that ships with Octopi comes preloaded with Python 2.7.16 and Python 3.7.3, hence why I tried specifying `pip3` and `python3` to prevent ambiguity.

bhlohumi, 2023-10-09:
I installed  this particular module as part of pipeline and it worked. Let me know if this helps.

MATN92, 2023-12-20:
this help 

sudo rm /usr/lib/python3.11/EXTERNALLY-MANAGED

amestsantim, 2023-12-30:
> this help
> 
> sudo rm /usr/lib/python3.11/EXTERNALLY-MANAGED

This worked for me

muava12, 2024-04-02:
> i get this error `AttributeError: module 'Cryptodome.Cipher.AES' has no attribute 'new'`

already resolved

bogukom, 2024-12-05:
had same issue 
```
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/pi/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
```
I have resolved my problem by 2 simple commands on pi:
```
sudo apt-get update
sudo apt-get install python3-pycryptodome
```
after that you can check if module has been installed properly:
`python3 -c 'from Cryptodome.Cipher import AES; print("Cryptodome works!")'`
The output should be:
`Cryptodome works!`

Then started flashing again and works fine