ct-Open-Source/tuya-convert #585: Gosund WB3 OTA flash problem

w4nnabe, 2020-04-11:
Hello there,

currently having trouble with a Gosund WB3 LED Bulb [https://templates.blakadder.com/gosund_WB3.html](url)

I bought 2 (both in one pack) of them a while ago, flashed 1 already successful. Yesterday i tried to flash the second one and something went wrong. For now the LED wont turn on anymore. No AP opened.
The LED came out of the box, was installed into the socket and started it's fast pairing mode blinking.

```Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Apr 10 22:45:10 2020 from 192.168.178.31
pi@raspberrypi:~ $ cd tuya-convert
pi@raspberrypi:~/tuya-convert $ dir
backups     docker      files              LICENSE    scripts
config.txt  Dockerfile  install_prereq.sh  README.md  start_flash.sh
pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.3
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 552.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 538.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
sudo: ufw: command not found
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.....................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
mkdir: cannot create directory ‘../backups/20200410_225516’: Permission denied
./start_flash.sh: line 96: pushd: ../backups/20200410_225516: No such file or directory
======================================================
Cleaning up...
Closing AP
Exiting...
```

Currently i have no clue how to fix it, cause there is no sign of life. 
I am far away beeing an expert on Linux, but looks like the flash didnt even start.
It failed to create the backup files already? 
Could it be a misstake made by me? A Device failure?
I would be thankful for any kind of help.

Thx

kueblc, 2020-04-11:
Could you post the rest of your logs please?

w4nnabe, 2020-04-11:
Thx. I am sry, thought it would be enough to attach the error msg in the terminal.

[smarthack-mqtt.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/4464087/smarthack-mqtt.log.txt)
[smarthack-psk.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/4464088/smarthack-psk.log.txt)
[smarthack-udp.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/4464089/smarthack-udp.log.txt)
[smarthack-web.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/4464090/smarthack-web.log.txt)
[smarthack-wifi.log.txt](https://github.com/ct-Open-Source/tuya-convert/files/4464091/smarthack-wifi.log.txt)

kueblc, 2020-04-11:
It failed to create the backup due to a permissions error. Looks like your device is still running the intermediate firmware.

You can `chown` the directory or just run as `sudo`. You do not need to connect a phone or put the device into pairing mode, since it already running the intermediate firmware. Just run `start_flash`, press enter, and plug the device back in.

w4nnabe, 2020-04-12:
Sry for the delay.

I ran it as sudo, but it fails to connect:

```root@raspberrypi:/home/pi/tuya-convert# sudo ./start_flash.sh
tuya-convert v2.4.3
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 547.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 534.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
sudo: ufw: command not found
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:31 --:--:--     0curl: (7) Failed to connect to 10.42.42.42 port 80: Connection timed out
Could not fetch a complete backup
Do you want to continue anyway? [y/N] N
======================================================
Cleaning up...
Closing AP
Exiting...
```

Seems the wifi.log updated:
[smarthack-wifi.log-2.txt](https://github.com/ct-Open-Source/tuya-convert/files/4465668/smarthack-wifi.log-2.txt)

So it was my fault, that i didnt run the command using sudo or run as root?
From my limited understanding, the backup is created before anything is uploaded?

ty

closed 2020-04-13 by w4nnabe