ct-Open-Source/tuya-convert #586: Teckin SB50 connects but smarthack packets never complete

stratus-ss, 2020-04-12:
### Before opening a new issue, please read the following

I just recently bought a pair of Teckin SB50 from Amazon. They connect to the tuya wifi but the smart packet section sends forever

Device name: Teckin SB50
Firmware Version: Unknown as I have never connected this to the app

I cloned a fresh copy of tuya convert (I have flashed a dozen smart plugs, some wifi leds and some power bars successfully) but I got the same result. I checked both of the mac addresses, but they appear compatible on the surface:

[image]

Host is Ubuntu 19.10 running from git not inside of a container

Logs attached:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4466614/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4466615/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4466616/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4466617/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4466618/smarthack-wifi.log)

kueblc, 2020-04-12:
Unfortunately these came with patched firmware, see #483

closed 2020-04-12 by kueblc

stratus-ss, 2020-04-12:
When a device is blocked, is there a chance that things will be worked around in the future?

I am not quite clear what "blocked" means.

kueblc, 2020-04-12:
Yes there is a chance a workaround will be developed, but nothing guaranteed

stratus-ss, 2020-04-12:
Thank you for your responses!

stratus-ss, 2020-04-18:
Do you accept bounties?