ct-Open-Source/tuya-convert #587: Flashed a non name power strip - Now web page wont come up

nitinuae, 2020-04-12:
So I flashed a no name strip (opened to see it was Tuya). Everything went well.
I had installed Kali Linux and the git cloned (today).

Now, the device come up (strange lights in the start - like a few flashes, then 12 flashes and then no light) with ESP-bdf788 as wifi id.

Connecting it I get 192.168.4.2.
Trying page 192.168.4.1 says refused to connect.

Its the same issue as issue 23 (which was closed). I have waited a few hours switching off and then on. Still I have same ESP-bdf788.

Do I have to take the switch apart and desolder the Tuya module to unbrick?

kueblc, 2020-04-12:
> So I flashed a no name strip (opened to see it was Tuya). Everything went well

You will want to ask your third party firmware provider for assistance with firmware configuration

nitinuae, 2020-04-12:
What is third party firmware here. I used tasmota.

kueblc, 2020-04-12:
We don't develop any of the third party firmware, you will need to seek support through Tasmota channels for post installation assistance.

closed 2020-04-12 by kueblc