ct-Open-Source/tuya-convert #588: Water leak sensor: Device did not appear with the intermediate firmware - related to #483?

kaybau, 2020-04-14:
Hello,
I tried several times to flash water leak sensor OTA with tyua-convert.
tuya-convert itself is working fine on my Raspi, I flashed succesfully other devices.
But for this one I got the message that the device did not appear with intermediate firmware. The device was blinking quickly in pairing mode and changed the blinking frequence once the flash program connected via VTRUST. In weblog I see it is the  ESP8266.
Here the logs:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4474387/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4474388/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4474389/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4474390/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4474391/smarthack-wifi.log)

If it is related to #483 - then you need luck in future to find  devices on market.
Cheers,
Kay

kueblc, 2020-04-14:
The good news: it does not look like #483, it is still using PSK format 01.

The bad news: it's not immediately clear what is causing the device not to take the upgrade.

kaybau, 2020-04-14:
[image]
This type is sold as different brands .
Using tuya software it connects as ESP-B62FC9 to my WLAN. I didn't dissamble it but it looks like the ESP chipset.

kaybau, 2020-04-20:
Hello,
this is the used chipset
[image]

Best,
Kay

kaybau, 2020-04-22:
Hi,
I could solve it like https://github.com/ct-Open-Source/tuya-convert/issues/135 
"the trick to keep them awake for flashing by changing the door contact state every few seconds (must be changed at least, less than every 6 seconds) by waving the magnet back and forth past the device."
For my device I bridged the water sensors every x seconds and got it updated.

closed 2020-04-22 by kaybau

MichaelMichaelMichaelMichaelMichael, 2020-11-19:
Same sensor labeled "WS06 Water Sensor", same chip, same results, I've tried with three of them, all three to not connect.
```
...
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
...........
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] ^C======================================================
Cleaning up...
No screen session found.
Closing AP
Exiting...

```

kueblc, 2020-11-19:
@MichaelMichaelMichaelMichaelMichael 

> Check the *.log files in the scripts folder

MichaelMichaelMichaelMichaelMichael, 2020-11-19:
Thanks for the hint, here's the logs on a fresh vanilla Raspi:
**smarthack-mqtt.log**
```
1605821656: mosquitto version 1.5.7 starting
1605821656: Using default config.
1605821656: Opening ipv4 listen socket on port 1883.
1605821656: Opening ipv6 listen socket on port 1883.
^C1605821776: mosquitto version 1.5.7 terminating
```
**smarthack-psk.log**
```
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
```
**smarthack-udp.log**
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C
```
**smarthack-web.log**
```
Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...
```
**smarthack-wifi.log**
```
System info
===========
94acc3b
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:15:a2:12 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 84:cc:a8:a5:2b:18
wlan0: AP-STA-CONNECTED 48:3f:da:70:3a:89
wlan0: AP-STA-DISCONNECTED 48:3f:da:70:3a:89
wlan0: AP-STA-DISCONNECTED 84:cc:a8:a5:2b:18
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```

MichaelMichaelMichaelMichaelMichael, 2020-11-19:
Ok, I got the secret, I had to press the button every two seconds and so I was able to flash them WS-06s. 
Anyone got a tasmota template for the WS-06?