ct-Open-Source/tuya-convert #590: No success flashing Kbaybo 300ml Aromatherapy Diffuser

imwlf, 2020-04-15:
Hey everyone. 
By browsing the Tasmota templates, i just stumbled on the Aroma Diffuser [](https://templates.blakadder.com/kbaybo_K-H25s.html). 
Is there anyone who has flashed the device yet or are there any instructions?
Thanks and best regards!

kueblc, 2020-04-15:
As manufacturers are constantly changing production methods and hardware, any given device model may not have the same chipset. In other words, having a Tasmota template is no guarantee that the device is still using an ESP82xx.

If you would like assistance diagnosing your flashing problem, you will need to upload your logs as requested in the instructions when opening a new issue. If you do not have a tuya-convert issue, please post your query elsewhere as this issue tracker is for tuya-convert issues only.

kueblc, 2020-06-20:
Closing until we receive more information

closed 2020-06-20 by kueblc