ct-Open-Source/tuya-convert #593: Magic Home Controller - Device did not appear with the intermediate firmware

snakejam, 2020-04-20:
Dear Team,

I try to flash my Magic Home WiFi Controller [found on Amazon](https://www.amazon.de/gp/product/B07GS1DM21/ref=ppx_yo_dt_b_asin_image_o06_s00?ie=UTF8&psc=1). 
The Controller is marked as "ZJ-WFMN-A V1.1", which should be a version of the ESP8285. I use a Raspberry Pi 4 with Raspbian Lite on it. 

When ./start_flash.sh is executed, I can connect to created AP vtrust-flash perfectly fine (needed to unblock all first). Then I put the connector into pairing mode by unplugging and replugging the power adapter three times and press Enter. After a while I get the message 
> Device did not appear with the intermediate firmware

But even when I study the log files, I have no clue why this isn't working:

## smarthack-wifi.log
```
System info
===========
31856ad
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:01:4e:f9 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 4c:66:41:4f:31:eb
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 4c:66:41:4f:31:eb
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```

## smarthack-web.log
```
Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...
```

## smarthack-udp.log
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C
```

## smarthack-psk.log
```
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
```

## smarthack-mqtt.log
```
1587419065: mosquitto version 1.5.7 starting
1587419065: Using default config.
1587419065: Opening ipv4 listen socket on port 1883.
1587419065: Opening ipv6 listen socket on port 1883.
^C1587419181: mosquitto version 1.5.7 terminating
```

I tried to do some research here, but in most cases, there is a kind of a connection from the ESP as seen in the logs. In my case, I only can find my mobile connecting to the AP. Is it possible that it has something to do with a new firmware on the ESP8285 like in #483?
Or did I do something horribly wrong on the setup? 

Many thanks in advance.

Kind regards,
Marcel

kueblc, 2020-04-20:
AFAIK Magic Home does not use Tuya modules or firmware. If you cannot use it with the Tuya SmartLife app, chances are it is not supported.

kueblc, 2020-04-20:
Closing as I can confirm this is not a Tuya product and thus incompatible with tuya-convert

closed 2020-04-20 by kueblc