ct-Open-Source/tuya-convert #594: Switch seems bricked.

estebanz01, 2020-04-22:
I tried to flash a generic wifi switch (sadly, I don't have a model/version) which is controlled via smart life app. The script detected correctly the device and tried to install the tasmota firmware, but the device ended up unresponsive and after a couple of minutes waiting for the process to finish, I stopped the script.

I tried to reconnect the power line on the device without success. It is still unresponsive and I'm not sure what else to do.

It's important to notice that the vtrust-recovery or the sonoff-**** SSID never appear on my phone.

Any help on this? thanks!
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4519397/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4519398/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4519399/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4519400/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4519401/smarthack-wifi.log)

kueblc, 2020-04-23:
Your logs are empty, can you check both `scripts` and `backups` directories for logs?

estebanz01, 2020-04-23:
weird. I'll post it here then. Here's the info that is located in the log files that lives within the `scripts/` folder:

mqtt
```
^C1587596356: mosquitto version 1.5.7 terminating
1587596464: mosquitto version 1.5.7 starting
1587596464: Using default config.
1587596464: Opening ipv4 listen socket on port 1883.
1587596464: Opening ipv6 listen socket on port 1883.
^C1587596770: mosquitto version 1.5.7 terminating
1587596895: mosquitto version 1.5.7 starting
1587596895: Using default config.
1587596895: Opening ipv4 listen socket on port 1883.
1587596895: Opening ipv6 listen socket on port 1883.
```
psk
```
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
```
udp
```
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
```

wifi
```
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED d8:f1:5b:bb:0f:bb
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
System info
===========
31856ad
Linux raspberrypi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
[... 147 lines cut ...]
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:19:99:25 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-CONNECTED 36:4a:8e:0c:99:32
wlan0: AP-STA-DISCONNECTED 36:4a:8e:0c:99:32
```

web
```
^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...
Listening on 10.42.
```

estebanz01, 2020-04-23:
then, there two folders within the backups folder:
```bash
pi@raspberrypi:~/tuya-convert/backups $ ls
20200422_230826  20200422_235528
```

For the `20200422_230826`, there's only two files:
```bash
pi@raspberrypi:~/tuya-convert/backups/20200422_230826 $ ls
device-info.txt  firmware-bb0fbb.bin
pi@raspberrypi:~/tuya-convert/backups/20200422_230826 $ cat device-info.txt 
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: bb0fbb
MAC: D8:F1:5B:BB:0F:BB
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
```

for the `20200422_235528` folder, there's a couple of smarthack logs:
```
pi@raspberrypi:~/tuya-convert/backups/20200422_235528 $ ls
device-info.txt  firmware-bb0fbb.bin  smarthack-mqtt.log  smarthack-psk.log  smarthack-udp.log  smarthack-web.log  smarthack-wifi.log
```

device-info:
```
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: bb0fbb
MAC: D8:F1:5B:BB:0F:BB
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
```

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4520196/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4520197/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4520198/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4520199/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4520200/smarthack-wifi.log)

CervDotBe, 2020-05-23:
Try to run tuya-convert again, it might be running the intermediate software. I had this as well. Switch looked bricked, until I restarted tuya-convert and the process picked up the device.

estebanz01, 2020-05-23:
I tried that a couple of times without success 😕