ct-Open-Source/tuya-convert #598: Device did not appear with the intermediate firmware - Teckin SL02 LED Strip

boggiano, 2020-04-28:
Hello,
I am trying to flash a [Teckin SL02 LED Strip ](https://templates.blakadder.com/teckin-SL02.html)
using a Pi (Raspbian GNU/Linux 10 - 4.19.97-v7+ #1294 SMP)

I can connect the tablet to the "vtrust-flash" AP but then I get  "Device did not appear with the intermediate firmware" message.

Any help, please ?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4544479/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4544482/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4544483/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4544484/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4544485/smarthack-wifi.log)

boggiano, 2020-04-28:
Sorry I tried again and again and it worked!

closed 2020-04-28 by boggiano

samabsalom, 2020-05-10:
hi there. how are you getting the LEDs into flash mode? 
also what GPIO pins are you using?
Thanks!

samabsalom, 2020-05-10:
for anyone that finds this. pulling the power five times got it into flash mode
GPIO 4 is red
GPIO 12 is green
GPIO 14 is blue