ct-Open-Source/tuya-convert #601: Issues with restore to original firmware

dal00, 2020-05-01:
Do I need to do something special to flash the original firmware back?

I have a device powered by an TYWE1S, did the full tuya-convert process successfully and got tasmota running on it. But tasmota was not understanding the second MCU device is not supported so I wanted to flash the original dump back.
I hooked up an RPI and flashed it back.. it is outputting stuff on serial so something is working but I'm not able to start the pairing by pressing the button for 5+5sec, led is not even flashing.

Unfortunatly I did not dump it via the serial before I started..

```
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: bbb045
MAC: 98:F4:AB:BB:B0:45
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1540c8
FlashChipRealSize: 2048K
Active Userspace: user2 0x81000
``

kueblc, 2020-05-01:
Connect to `vtrust-recovery` and browse to http://10.42.42.42/undo

dal00, 2020-05-01:
Is that served by the esp? Where should that ssid be?

kueblc, 2020-05-01:
`vtrust-recovery` is served by the ESP once you flash the backup to the device.

`upgrade.bin` is the intermediate firmware that is first uploaded to the device in order to create the backup and flash custom firmware.

dal00, 2020-05-01:
I don't see that network. It is not there as a hidden ssid either. Do I need to do anything else for it to show up?