ct-Open-Source/tuya-convert #602: Device did not appear with the intermediate firmware Blitzwolf & gosund switches

inv1sible, 2020-05-01:
First thank you for generating this issue to me. Although I am dissappointed not being able to flash the 8 switches laying around here I learned a lot new from docker, screen and ubuntu in general.

I also read the others issues to the same outcome but did not succeed so here are my parts:

Affected models: 
* gosund SP111 15A (bought 04/2020 from amazon), 
* Blitzwolf SHP6 15A (new version SP111 newer layout no more adapter USBART flashing :cry:)

Logs:
[image]

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4566385/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4566386/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4566387/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4566388/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4566389/smarthack-wifi.log)

Hoping anyone has a hint to me. THANKS IN ADVANCE

ortegafernando, 2020-05-02:
Hi, i think you have same problem as me. See this:

[](https://github.com/ct-Open-Source/tuya-convert/issues/483)

Tuya "has changed" their software, so tuya-convert has to "investigate" about it.

inv1sible, 2020-05-03:
Thank you @ortegafernando for this information. Sounds logical and thats how it always is when trying to use creative functions i.e. exploiting things...

@kueblc is there already so working on this? How can I help investigate?

weidlix, 2020-05-04:
I have the same issue with a Sonoff TH16. Any news on this issue?

PaulMerk, 2020-05-06:
I have the same issue with Gosund SP112

ThePirat91, 2020-05-12:
@PaulMerk I noted this too on all my (new) bulbs....different behavior from the previous firmware version...

kueblc, 2020-05-12:
@inv1sible unfortunately I'm not quite sure what's causing this issue for you, it is not #483 as you do not have a PSK identity starting with `02`. I can see in your wifi log that an Espressif device has connected, but it disconnected before making any requests.

You could help debug this by doing a network recording while attempting tuya-convert. You can use WireShark or `tcpdump` to do this, ie. `tcpdump -i <your wireless interface> -w capture.pcap`.

@weidlix tuya-convert only works with Tuya devices, not Sonoff.

@PaulMerk @ThePirat91 this just means smart config has completed successfully. To debug further, you must check your logs.

inv1sible, 2020-05-12:
Thx @kueblc for your reply. As you asked for I did record the network traffic. From my quick review in WS I saw that the AP tries to connect to 10.42.42.29 instead of 10.42.42.42 that I saw hardcoded a lot in tuya-convert. So changed to ...29 in the scripts (sorry if that's stupid) but at least it got somewhere... not doing a backup and not transferring but maybe this helps?  Looking out for a reply... THX in advance

[Archiv.zip](https://github.com/ct-Open-Source/tuya-convert/files/4618191/Archiv.zip)

kueblc, 2020-05-14:
You should not change the hardcoded IP in the script, it needs to be `10.42.42.42` for it to function properly. This is used by the intermediate firmware to tell the script that it is ready to proceed to flashing. Changing it only tricks the script into thinking the device is ready when it is not.

Other than that, your second recording looks good. I can see the device authenticating. You should be able to find evidence of this in your psk log. I would try again without the script modification and see if you get any requests appearing in your web log.

inv1sible, 2020-05-14:
Thanks @kueblc for the explanation. "10.42.42.42 .... This is used by the intermediate firmware to tell the script that it is ready to proceed" ... sounds fair... Thats why I initially reverted all my changes and recorded first capture.pcap.

The second recording (capture-29.pcap) is the result of ./start-flash.sh after changing to 10.42.42.29... 

Or did I got you wrong and you mean capture.png looks good?

lumavp, 2020-05-26:
I have the same problem with a smart plug Bakibo T22Y. Tried for several hours, even reinstalling raspberry pi and installing tuya donor on a NodeMCU.  The device disconnects and exits config mode.
Same output on wifi log as above.

Noschvie, 2020-06-05:
Yesterday I successfully flashed a SP112 device. SP112 was brand new and never connected to the internet before.
Thanks for tuya-convert !

inv1sible, 2020-06-07:
I was able to flash the power plugs (Blitzwolf and Gosund) using a raspbian stretch with tuya-convert v2.4.3. No changes needed, just worked. 

So I assume my three wifi adapters have issues in providing the AP under ubuntu 20.04.

tldr; Raspberry Pi 3b, raspbian stretch, tuya-convert v.2.4.3 WORKED.

closed 2020-06-07 by inv1sible