ct-Open-Source/tuya-convert #603: LE lampUX light bulbs fail to flash

Draxter, 2020-05-02:
Hi, Having trouble with a light bulb that is on the list marked as working with tuya-convert here: https://templates.blakadder.com/le_901001-EU-a.html

Getting a 'Device did not appear with the intermediate firmware'. Versions and logs below:

Model: LE lampUX 8.5W 806lm RGBCCT Bulb
Tuya-Convert version: 2.4.3
Tuya Firmware version: 3.4.30

smarthack-mqtt.log(includes some random calls to my other MQTT devices, ignore these):
https://pastebin.com/cf6wWpft

smarthack-psk.log:
https://pastebin.com/8LKHF8Wz

smarthack-udp.log:                                                                                                                                                                                                                                                      
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
192.168.1.149 {"ip":"192.168.1.149","gwId":"84325045d8f15be22ec9","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"keyj979nf3q3theh","version":"3.3"}
192.168.1.170 {"ip":"192.168.1.170","gwId":"843250452462ab4ea02e","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"keyj979nf3q3theh","version":"3.3"}
192.168.1.9 {"ip":"192.168.1.9","gwId":"47163522cc50e3e660c8","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"IAYz2WK1th0cMLmL","version":"3.3"}

smarthack-web.log:
Listening on 10.42.42.1:80
[I 200502 07:59:05 web:2250] 302 GET /hotspot-detect.html (10.42.42.39) 4.07ms
[I 200502 07:59:05 web:2250] 200 GET / (10.42.42.39) 4.94ms
[I 200502 07:59:09 web:2250] 302 GET /hotspot-detect.html (10.42.42.39) 2.91ms
[I 200502 07:59:09 web:2250] 200 GET / (10.42.42.39) 3.17ms
[I 200502 07:59:09 web:2250] 302 GET /hotspot-detect.html (10.42.42.39) 3.22ms
[I 200502 07:59:09 web:2250] 200 GET / (10.42.42.39) 4.15ms
^CReceived SIGINT, exiting...

smarthack-wifi.log:
https://pastebin.com/aKqw7Gav

kueblc, 2020-05-02:
Unfortunately you have patched firmware

Duplicate of #483

closed 2020-05-02 by kueblc

Draxter, 2020-05-02:
@kueblc Is tuya-convert likely to develop a patch to allow users to flash this new firmware or should I try to get lightbulbs with the older firmware?