ct-Open-Source/tuya-convert #605: Please add some comments regarding Tasmota

barbudor, 2020-05-03:
Hi
I have no issues with Tuya-convert and I have used quite a few times to convert some devices.
Many thanks for this :+1:

I am an active user of Tasmota and helping people on the Tasmota Discord  server and we often face people that are trying to be smart but bricks their devices because they replace the `tasmota.bin` file that you provide with the unappropriate `tasmota-minimal.bin`.

May I kindly ask the following :
- Stick with the original name `tasmota-wifiman.bin` (instead of renaming to `tasmota.bin`), that might help people understand they should use this binary and not a random one
- Add a comment to make clear that if someone wants to change the provided firmware by a more recent they should use `tasmota-wifiman.bin` or get help on the Tasmota Discord server.
(I can do a PR on the README.md if you like)

Thanks again for your hard work, this is great to be able to really own those Tuya devices.

Best regards to everyone
#keepsafe

andre68723, 2020-05-22:
thanks @barbudor that would save others
your description is actually how I bricked one device, trying to save some time by replacing the bin

kueblc, 2020-05-22:
Thanks for the suggestions @barbudor. If you'd like to make a PR against `development`, I will gladly include it in the next release.

barbudor, 2020-05-24:
Hi @kueblc 
Done as https://github.com/ct-Open-Source/tuya-convert/pull/641
Hopefully this will reduce number of people bricking their devices.
Thanks

closed 2020-05-24 by barbudor

barbudor, 2020-05-24:
Apologies for the messy PR.
The right PR against development is https://github.com/ct-Open-Source/tuya-convert/pull/642
I haven't found a way to remote PR 641
Thanks