ct-Open-Source/tuya-convert #606: (SP111 v1.1) Device did not appear with the intermediate firmware

delusho, 2020-05-03:
issues getting a device to flash

Hello - I am amazed by all the great work and love which went into tuya-convert. Super!

I ordered 4 Gosund SP111 v1.1 from amazon late in April and have managed to flash three of them with tuya convert. However, I am having trouble with the last one while sending the SmartConfig Packets.

I've looked through my logs and couldn't find anything comparable to other related issues here.

The device is in paring mode with the LED rapidly flashing. After the attempt the LED continues to flash slowly -  I am able to put the outlet into pairing mode again by long-pressing the button.
It would be great if someone could help me!

> > pi@raspberrypi:~/tuya-convert $ sudo ./start_flash.sh
> tuya-convert v2.4.3
> Checking for network interface wlan0... Found.
> Checking UDP port 53... Available.
> Checking UDP port 67... Available.
> Checking TCP port 80... Available.
> Checking TCP port 443... Available.
> Checking UDP port 6666... Available.
> Checking UDP port 6667... Available.
> Checking TCP port 1883... Available.
> Checking TCP port 8886... Available.
> sudo: ufw: command not found
> ======================================================
>   Starting AP in a screen
>   Starting web server in a screen
>   Starting Mosquitto in a screen
>   Starting PSK frontend in a screen
>   Starting Tuya Discovery in a screen
> 
> ======================================================
> 
> IMPORTANT
> 1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
>    This step is IMPORTANT otherwise the smartconfig may not work!
> 2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
>    Make sure nothing else is plugged into your IoT device while attempting to flash.
> 3. Press ENTER to continue
> 
> 
> ======================================================
> Starting smart config pairing procedure
> Waiting for the device to install the intermediate firmware
> ..Put device in EZ config mode (blinking fast)
> Sending SSID                  vtrust-flash
> .Sending wifiPassword          
> Sending token                 00000000
> Sending secret                0101
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ...............
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> ......................
> Device did not appear with the intermediate firmware
> Check the *.log files in the scripts folder
> Do you want to try flashing another device? [y/N] n
> ======================================================
> Cleaning up...
> Closing AP
> Exiting...
> pi@raspberrypi:~/tuya-convert $ 
> 

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4571238/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4571240/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4571241/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4571242/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4571243/smarthack-wifi.log)

TheTimmaeh, 2020-05-03:
I had the exact same issue with the SP112 not being able to convert 4 plugs after sucessfully converting 2. After lots of research I simply started from scratch setting up tuya-convert again etc. and it magically worked.

barbudor, 2020-05-03:
Hi @delusho 
I have successfully flashed 2 series of 4x Gosund SP111.
I had some difficulties with the 2nd series with similar messages as yours in the smart-psk.log
I used the "development" branch instead of "master" and it worked better.
May be that could help
Cheers

delusho, 2020-05-05:
Thanks @TheTimmaeh @barbudor I tried both and it didn't work.

barbudor, 2020-05-05:
@delusho SP111 are easy to open and there are flashing spots easily accessible.
There is a screw hidden in the earth hole on the plug side.

ortegafernando, 2020-05-06:
https://github.com/ct-Open-Source/tuya-convert/issues/483

TheTimmaeh, 2020-05-07:
> #483

I don't see any signs of this in the psk log file.

ugomeda, 2020-05-14:
> I had the exact same issue with the SP112 not being able to convert 4 plugs after sucessfully converting 2. After lots of research I simply started from scratch setting up tuya-convert again etc. and it magically worked.

Thanks for this tip !

I couldn't flash my device, same error. I did an `rm -rf` of tuya-convert and started fresh. It worked ! Sadly i did not keep the initial logs to help on this issue.

kueblc, 2020-05-21:
@delusho unfortunately it looks like the model you received does not use an ESP82xx microcontroller and so is not supported by tuya-convert or alternative firmwares such as Tasmota or Espurna.

pmax3, 2020-05-22:
Hi all, just wanted to tell that I had the same issue with recently purchased Gosund SP111 v1.1 plugs. Both of them showed the message "device did not appear with the intermediate firmware". 
First I tried to reload the master branch without success, but later I tried with the development branch and it worked. While working with the development version the 2nd device did not start with the fast blinking LED and I was not able to start this provisioning mode while pressing the button on the device. I gave it a try and started the flashing process anyway and it seemed that this devices had booted already into the intermediate firmware and the script loaded the tasmota.bin without any problems.

lukasvermeer, 2020-05-28:
> unfortunately it looks like the model you received does not use an ESP82xx microcontroller

Could you please explain how you deduced that from these logs, @kueblc?

I suspect I have the same issue (I managed to flash three plugs from the box ordered through Amazon, but cannot seem to flash the fourth.)

lukasvermeer, 2020-05-28:
> Hello - I am amazed by all the great work and love which went into tuya-convert. Super!

+100 on this, btw. I am really impressed by how easy this was to use. 👍

kueblc, 2020-05-28:
@lukasvermeer there are three possible clues, it may print a warning in the psk or web logs, or you can check the [MAC address OUI](https://www.wireshark.org/tools/oui-lookup.html) and see if the manufacturer is "Espressif Inc".

Happy to hear you had an easy time with this tool! Feel free to let me know any thoughts on improvements.

lukasvermeer, 2020-05-28:
@kueblc Useful pointer. Thanks! I see only one MAC address in the logs, and that's my phone.

lukasvermeer, 2020-05-28:
Managed to pull the development branch in the docker and flash successfully. Thanks for all the help. Logs below in case it helps anyone else.

```
$> docker exec -it lukas_tuya_1 bash
root@Pip:/# cd /usr/bin/tuya-convert/
root@Pip:/usr/bin/tuya-convert# git checkout development
M	config.txt
Switched to branch 'development'
root@Pip:/usr/bin/tuya-convert# git pull origin development
From https://github.com/ct-Open-Source/tuya-convert
 * branch            development -> FETCH_HEAD
Already up to date.
root@Pip:/usr/bin/tuya-convert# ./install_prereq.sh 
Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main Sources [408 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main armhf Packages [803 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe armhf Packages [1,154 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports bionic-security/main Sources [192 kB] 
Get:9 http://ports.ubuntu.com/ubuntu-ports bionic-security/main armhf Packages [533 kB]
Fetched 3,341 kB in 2s (2,215 kB/s)                      
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
iw is already the newest version (4.14-0.1).
net-tools is already the newest version (1.60+git20161116.90da8a0-1ubuntu1).
python3-setuptools is already the newest version (39.0.1-2).
dnsmasq is already the newest version (2.79-1).
haveged is already the newest version (1.9.1-6).
python3-wheel is already the newest version (0.30.0-0.2).
curl is already the newest version (7.58.0-2ubuntu3.8).
git is already the newest version (1:2.17.1-1ubuntu0.7).
python3-dev is already the newest version (3.6.7-1~18.04).
screen is already the newest version (4.6.2-1ubuntu1).
hostapd is already the newest version (2:2.6-15ubuntu2.5).
mosquitto is already the newest version (1.4.15-2ubuntu0.18.04.3).
python3-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.1).
The following additional packages will be installed:
  libssl1.1
Suggested packages:
  libssl-doc
The following packages will be upgraded:
  libssl-dev libssl1.1
2 upgraded, 0 newly installed, 0 to remove and 49 not upgraded.
Need to get 2,436 kB of archives.
After this operation, 1,024 B of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main armhf libssl-dev armhf 1.1.1-1ubuntu2.1~18.04.6 [1,367 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main armhf libssl1.1 armhf 1.1.1-1ubuntu2.1~18.04.6 [1,069 kB]
Fetched 2,436 kB in 0s (11.8 MB/s)
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 2.)
debconf: falling back to frontend: Readline
Preconfiguring packages ...
(Reading database ... 19109 files and directories currently installed.)
Preparing to unpack .../libssl-dev_1.1.1-1ubuntu2.1~18.04.6_armhf.deb ...
Unpacking libssl-dev:armhf (1.1.1-1ubuntu2.1~18.04.6) over (1.1.1-1ubuntu2.1~18.04.5) ...
Preparing to unpack .../libssl1.1_1.1.1-1ubuntu2.1~18.04.6_armhf.deb ...
Unpacking libssl1.1:armhf (1.1.1-1ubuntu2.1~18.04.6) over (1.1.1-1ubuntu2.1~18.04.5) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Setting up libssl1.1:armhf (1.1.1-1ubuntu2.1~18.04.6) ...
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76.)
debconf: falling back to frontend: Readline
Setting up libssl-dev:armhf (1.1.1-1ubuntu2.1~18.04.6) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Collecting git+https://github.com/drbild/sslpsk.git
[... 76 lines cut ...]

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  q) quit; do nothing
Please select 0-2: 2
Are you sure you want to flash tasmota.bin? This is the point of no return [y/N] y
Attempting to flash tasmota.bin, this may take a few seconds...
Flashed http://10.42.42.1/files/tasmota.bin successfully in 16328ms, rebooting...
Look for a tasmota-xxxx SSID to which you can connect and configure
Be sure to configure your device for proper function!

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
```

kueblc, 2020-05-28:
Awesome. Happy to hear it!

closed 2020-05-28 by kueblc