ct-Open-Source/tuya-convert #607: Link in readme no longer works (404)

nickdos, 2020-05-03:
In the readme, there is a link to an image: 

https://images.offensive-security.com/arm-images/kali-linux-2018.4a-rpi3-nexmon-64.img.xz

This link no longer resolves (throws 404), so please either update the link or remove the reference to it.

PoloB12, 2020-05-04:
You can also use a standard Rapsbian image, no need for Kali

nickdos, 2020-05-04:
I used the image above (v 2020.1a) in the end and it worked perfectly the first time (rPi 3B+).

I had previously used Raspbian and the installation step did not go well (errors/conflicts that required intervention via apt-get, etc.). I also had problems getting the network working properly.

MikeDProducer, 2020-05-11:
I can't figure out what the default passwords are for this kali build.

nickdos, 2020-05-11:
From memory its `toor` (root backwards).