ct-Open-Source/tuya-convert #609: vtrust-flash wifi network not showing up

wil-sys, 2020-05-07:
i was trying to flash my tuya smart plug with tasmota i ran start_flash.sh as root and when i got to the part where it said to connect to the wifi network the network did not show up 

attached are my smarthack wifi log
smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4595056/smarthack-wifi.log)

kueblc, 2020-05-09:
`lo` is your loopback interface, you cannot start an access point on it because it is not a WiFi device. You must set your WiFi interface in the configuration.

MikeDProducer, 2020-05-11:
I am having this issue also. Very strange because it worked before and I am using the same steps.

dragonpiper, 2020-05-16:
Also have the same issue. unblock also didn't work

`Linux amd-server 4.15.0-99-generic #100-Ubuntu SMP Wed Apr 22 20:32:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
OpenSSL 1.1.1  11 Sep 2018
Dnsmasq version 2.79  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.6
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2016, Jouni Malinen <j@w1.fi> and contributors
Python 3.6.9
===========
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr a0:04:60:1b:b8:92 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED`

panuruj, 2020-05-30:
I had this issue, and I fixed it by setting the country for the wifi adapter.
* `sudo raspi-config`
* Localisation Options
* Change wireless country

forall, 2020-06-09:
Which country do you set, because I still  have problem after run ./start_flash.sh

pi@raspberrypi:~/tuya-convert $ ./start_flash.sh 
tuya-convert v2.4.3
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

======================================================
Cleaning up...
No screen session found.
No screen session found.
No screen session found.
No screen session found.
Closing AP
Exiting...

panuruj, 2020-06-09:
> Which country do you set, because I still have problem after run ./start_flash.sh
>
> # pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
> tuya-convert v2.4.3
> TUYA-CONVERT [...]

USA works for me.

forall, 2020-06-24:
I set USA country but I have the same problem.