ct-Open-Source/tuya-convert #612: New Tuya firmware from last month (SmartLife, firmware 1.6.1 on Teckin 13w bulbs)

vsaintloup, 2020-05-10:
Hello, newbie here. 

I successfully used tuya-convert for all of my Tuya devices at home, although I had 4 out of 5 of some nice 13W bulbs ( https://www.amazon.com/TECKIN-Changing-Equivalent-Compatible-2900K-6000K/dp/B07RZKV7FQ/ ) were recently updated to SmartLife (tuya)'s new firmware last month version 1.6.1, so for those I had to access the PIN inside the lights in order to flash the firmware since they refused to go further than "resending packets" in the OTA process.

I am on Mac, I used the Tazmotizer app, and I did a backup of that new firmware thinking that it might help the owner of this repo to find a way to update the OTA tool in response to this new tuya patched firmware… 

If this can be of any help, just let me know where Tazmotizer stores its back-ups (I can't find this information on Google lol) and I'll send it to you! 

Thanks for your amazing work!

vsaintloup, 2020-05-10:
Also, in case anyone is looking for the correct GPIO output for those powerful 13w RGBWW Teckin bulbs, here the correct config:

 ```
esphome:
  name: teckin_13w_rgbww
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "AirPort 2.4 GHz"
  password: "0000000000"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Light 13W Hallway"
    password: "0000000000"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "0000000000"

ota:
  password: "0000000000"

output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO4
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO12
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO14
  - platform: esp8266_pwm
    id: output_warm_white
    pin: GPIO13
  - platform: esp8266_pwm
    id: output_cold_white
    pin: GPIO5

# Define a light entity
light:
  - platform: rgbww
    name: teckin_13w_rgbww
    id: teckin_13w_rgbww
    red: output_red
    green: output_green
    blue: output_blue
    warm_white: output_warm_white
   cold_white: output_cold_white
    cold_white_color_temperature: 6200 K
    warm_white_color_temperature: 2800 K
```

MikeDProducer, 2020-05-11:
I tried to flash this bulb OTA today and never got
It working. You were able to make it work by flashing the hardware directly? May I ask these questions:

Did you have to solder?

Also how did you get the top off the bulb?

I have soldering equipment. I just want to be prepared. Thank you for the info you’ve already shared here! It’ll be helpful!

vsaintloup, 2020-05-12:
Hello @MikeDProducer ,

No I did not solder; so… my method was risky! I was ready to lose/brick a bulb or two in the process… but turns out I was able to do all of my 5 bulbs without any problem! The only one that worked OTA was a brand new one that I had in a box since a few months (so it had the previous firmware).

Here's how I did it, it will leave a very minimal & cosmetic dent:
[image]

Use a flat tool and hit it with the palm of your hand, the cover will come right off
[image]
[image]

Then, remove the 4 screws in order to remove the LEDs
[image]

Once the LED are removed, look at the GPIOS:
[image]

You'll need some patience to build yourself a little tool like this in order to connect it to your USB to serial adaptor:
[image]
[image]

Now the tricky part, GPIO 0 needs to be connected to the ground output of your serial adaptor  for a few seconds when powered up to put it in flash mode. My serial adaptor had a second ground connector (for the 5V, so I just used this one with a jumper cable). And here's how I did it: 1) get my Tazmotizer ready to fire, 2) plug the USB to serial adaptor, 3) with left hand you make the contact between GPIO 0 & ground, 4) with right hand you place your connector on the other pins, 5) wait 2 or 3 seconds, 6) the device is in flash mode, you can remove your left hand from maintaining ground contact to GPIO 0 and use it to click «Tazmotize!», now hold steady and don't move… (the upload process is super fast if you skip backing up the firmware, backing it up takes forever PLUS I wasn't able to flash after backing it up, I had to do a power-cycle! Doing power-cycles while holding down this cables in this tiny space is no fun lol) 
[image]

More risks than soldering, but hey it took me less than 2 minutes per bulb!

Lerrpy, 2020-05-12:
👏👏👏

MikeDProducer, 2020-05-12:
Wow. This is the greatest reply I have ever received on a forum. Ever. You, sir, are magnificent!

MikeDProducer, 2020-05-12:
It seems like I can only get bulbs to work OTA if they are new and in the box these days.

vsaintloup, 2020-05-12:
Yea, well probably because they stayed in warehouses… whereas the firmware of the ones that we are using was wirelessly and automatically updated. I can remember a few weeks ago that I got a notification from the SmartLife app saying «Firmware successfully updated!» (without me asking for it, obviously)… their notion of consent is simply non-existent lol

MikeDProducer, 2020-05-12:
Any tricks to getting it back together? Haha these pins look very tiny. The 8 pins.

vsaintloup, 2020-05-12:
Yes indeed they are! Upside down with your cellphone flashlight on your desk can help…

MikeDProducer, 2020-05-12:
Thank you! I need to find those little pins like you used to connect to the leads. I just ordered some. I tried to solder it but it’s such a narrow opening.

vsaintloup, 2020-05-12:
@MikeDProducer I got those pins from when ordering some Wemos D1 Minis… I don't know how much you are about to pay for those pins, but maybe you could have some D1 Minis for a few more dollars 😅 https://www.amazon.com/IZOKEE-NodeMcu-Internet-Development-Compatible/dp/B076F52NQD

MikeDProducer, 2020-05-12:
Thanks for your help. I can’t begin to tell you how helpful this was. I’ve been staring at that bulb all day considering taking a hammer out of frustration. Haha.

wzaatar, 2020-05-16:
@vsaintloup, thanks for the write-up. If I understand correctly, no OTA for this firmware as of today?

vsaintloup, 2020-05-16:
Nop, the 1.6.1 version on these Teckin 13W can not be flashed with tuya-convert OTA (assuming there hasn’t been a new tuya-convert version addressing that issue in the last 7 days). 

These 13W bulbs did not originally shipped with 1.6.1, so any of those bulbs that were not connected to the WiFi since about a month or two will (including many brand new ones that were sleeping on a shelf in a warehouse since many months) have the previous firmware on which Tuya-convert works like a charm.

TripitakaBC, 2020-12-09:
@vsaintloup 
Nice post! I just got a set of 4 SB53 delivered from Amazon and found out they wouldn't OTA flash. That's when I found this post. I'm not having the same success that you did; I've made the same rig and I'm following your guide but either my USB>Serial isn't the correct one or the manufacturer has done something to the bulbs to prevent flashing.

I'm using an ESP Link v1.0 USB to UART but when I hit 'Tasmotize!', I get no progress bars and then a time-out failure. What USB>Serial did you use?

TripitakaBC, 2020-12-09:
Ah...a little persistence paid off.
That ESP Link just didn't have the stones to power the upgrade. I sat and considered the situation and figured that I had a few Wemos D1 minis, a couple of ESP32's, an Arduino Uno and a couple of RPis kicking around that may work as a makeshift USB to Serial with more juice.

I went with the Arduino Uno, simply because it was the easiest to implement but I still couldn't get the contacts to work and I still wasn't wholly convinced that the issue wasn't with the firmware or hardware itself. I resorted to soldering wires in place on the board and...it worked!

So, just to clarify on the excellent instructions above, the GPIO0 wire needs to be in place BEFORE power is applied to the 3.3v connection. I found that I could leave it connected the whole time and it flashed OK.

I've done many ESPhome devices before but this is my first tango with Tasmota so I had to read up on what to do next. That was straightforward but I did find this [excellent resource](https://templates.blakadder.com/teckin_SB53.html) that had the correct template for the SB53. That template goes under 'Configuration' and then 'Configure Other'. Not sure I would have figured it out without that.

Many thanks @vsaintloup - you are todays superstar, to me at least.

ghajj, 2021-02-18:
> 
> 
> > Use a flat tool and hit it with the palm of your hand, the cover will come right off
> 
> Smash it with your hand against a rug on the floor and it will pop off the top without having to use a tool to mar the plastic.

This worked. The cover popped up. Thx

codyjamestechnical, 2021-03-02:
@vsaintloup you should set color_interlock = true  and constant_brightness = true in your config to prevent overheating and damage to the board.

pritam2016bakshi, 2022-01-21:
Can you give me original firmware(**bulb**).?