ct-Open-Source/tuya-convert #613: Socket won't flash

leondeklerk, 2020-05-12:
### Before opening a new issue, please read the following

Hi all,

I received my new socket/plug today. And hoped to flash it with Tasmota as I did with the last one. 
Unfortunately, this was not the case, as flashing won't work. 

As is usual with AliExpress I'm not exactly sure what model this is. 
The box itself has a label WS-UEU16-M and on the bottom of the device, JH-G01E is written. And the tuya smart app calls it 'BS Smart Socket with Power Meter' by default. 
Firmware:
- Wi-Fi Module: 1.0.6
- MCU Module: 1.0.6

Logs in /backups
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4617878/smarthack-mqtt.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4617880/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4617882/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4617884/smarthack-wifi.log)

Log in /scripts
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4617886/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4617887/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4617890/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4617891/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4617892/smarthack-wifi.log)

I hope someone can help me out, thanks in advance.

Console output: 
```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (7) Failed to connect to 10.42.42.42 port 80: Connection refused
Could not fetch a complete backup
Do you want to continue anyway? [y/N] y
======================================================
Getting Info from IoT-device
======================================================
Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
  Tasmota v8.1.0.2 (wifiman)
  ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  q) quit; do nothing
Please select 0-2: 2
Are you sure you want to flash tasmota.bin? This is the point of no return [y/N] y
Attempting to flash tasmota.bin, this may take a few seconds...
Could not reach the device!
```

kueblc, 2020-05-13:
I only see one device in your wifi log. Can you identify it?

leondeklerk, 2020-05-13:
> I only see one device in your wifi log. Can you identify it?

Yes this is my phone (Oneplus 7 Pro), which I connected by setting IP settings to static instead of DHCP. I gave it a random IP since DHCP left my phone on "obtaining IP address".

kueblc, 2020-05-21:
Is your phone assigned `10.42.42.42`? That address must not be used as it is reserved for a device running our intermediate firmware.

leondeklerk, 2020-05-21:
No I gave it IP 192.168.1.128, which is not even assigned on my local network.

kueblc, 2020-05-21:
Hmm puzzling, as I only see one device in your wifi log. Do you know what this is then?

> IoT-device is online with ip 10.42.42.42

leondeklerk, 2020-05-21:
I must admit that I have no clue, there is nothing in my own network on the 10.xxx.xxx.xxx format, so I don't think it would be a random device. The only thing I can think of is the smart socket itself, but I don't know how all that works.

marji, 2020-05-27:
I just want to add to the above "mystery" that I think the 10.42.42.42 in the logs is actually the socket/plug, stuck/half flashed.

I found this github issue here while searching because I failed to flash my own smart plug device (Brilliant Smart WiFi Plug with USB Charger) - see the connection timeout:

```IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
Failed to connect to 10.42.42.42 port 80: Connection timed out
```
So it is online, but it is not listening on port 80.
Then, when I try to flash my socket:
```Attempting to flash firmware.bin, this may take a few seconds...
Could not reach the device!
```
while I could ping the device:
```pi@air:~/tuya-convert $ ping 10.42.42.42
PING 10.42.42.42 (10.42.42.42) 56(84) bytes of data.
64 bytes from 10.42.42.42: icmp_seq=1 ttl=128 time=5.98 ms
64 bytes from 10.42.42.42: icmp_seq=2 ttl=128 time=6.05 ms
```
and when I unpowered it it, it stopped replying (a proof it was the device on this IP address):
```64 bytes from 10.42.42.42: icmp_seq=13 ttl=128 time=2.97 ms
From 10.42.42.1 icmp_seq=51 Destination Host Unreachable
```

BTW. After I powered the smart plug again, it started listening on port 80 and I was able to finish the flashing successfully!

leondeklerk, 2020-05-27:
> BTW. After I powered the smart plug again, it started listening on port 80 and I was able to finish the flashing successfully (while actually typing this update on this ticket! :)
@leondeklerk Perhaps try to remove your socket from power, wait a bit, power it again and try to flash again.

I planned on using this tactic, but this time I connected with another mobile phone. This time it did connect instead of staying on "obtaining ip" and everything went trough! Now my plug is flashed and should work good :) (when I figure out the pins). Thanks for the help

leondeklerk, 2020-05-27:
Btw is there maybe a way to decode the backup firmware file it created? That might actually help me out finding the correct pins.

kueblc, 2020-05-28:
Happy to hear both of your flashing attempts finally succeeded!

As far as using the firmware, unless you are very skilled with a disassembler I don't think you will find it useful for determining pin function.

If the Tuya firmware was configuration based like Tasmota, I would love to automatically generate sensible defaults, but it is written and compiled by each manufacturer individually.

Closing as the original issue has been resolved. Let me know if you have any other concerns.

closed 2020-05-28 by kueblc

leondeklerk, 2020-05-28:
Ah okay, thanks anyway. For the support and everything you do for this project!

miaszh, 2020-09-18:
Hi
I actually faced the same issue while flashing which leaded me to flash remove and retry and only then it was flashed successfully.
However I cannot find any working template which indicates the Power meter, could you find any?

Thanks

leondeklerk, 2020-09-18:
> Hi
> I actually faced the same issue while flashing which leaded me to flash remove and retry and only then it was flashed successfully.
> However I cannot find any working template which indicates the Power meter, could you find any?
> 
> Thanks

Hi,

I found the pin layout together with someone called Niko (of whom I unfortunately do not have the Github username, so couldn't tag here). We found the template to be this one:
[https://templates.blakadder.com/avatto_JH-G01E.html](url)

Hope that one also works for you!

miaszh, 2020-09-18:
Yes! Thats it.
Thank you!