ct-Open-Source/tuya-convert #614: Wall plug dead after flashing with plugged in consumer device? (Blitzwolf BW-SHP8)

waltner, 2020-05-13:
Hi,

I have started to add smart devices to my home and flashed some with tuya-convert already. Thank you for the great work!

Today I wanted to flash the **Blitzwolf BW-SHP8** and after installing it into the wall I checked if power was available at the washing machine. It was so I started to flash but forgot to unplug the wm.

During the flashing procedure the LED went off and since then the device seems to be dead. Here is the log:
```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
..................
SmartConfig complete.
Resending SmartConfig Packets
......./start_flash.sh: line 129: 14751 Terminated              ./smartconfig/main.py
............................................................................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  37671      0  0:00:27  0:00:27 --:--:-- 32840
curl: Saved to filename 'firmware-c87b83.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: c87b83
MAC: 2C:F4:32:C8:7B:83
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================
Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
  Tasmota v8.1.0.2 (wifiman)
  ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  q) quit; do nothing
```

You can see the available options were limited. ;)

Logs in backup:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4621453/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4621455/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4621456/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4621457/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4621460/smarthack-wifi.log) [`24:df:6a:34:02:85` is my mobile]

As there was no more power and the LED did not light up upon button press, I removed the socket from the wall. Then I soldered the 4 pins (`GND, 3.3V, RX, TX`) on the `TYWE2S` chip and connected them to my `CH340G` device (RX <-> TX, TX <-> RX). I also checked the connectivity with a multimeter.
After holding the button pressed before connecting the `CH340G` to my Laptop and releasing the button after some seconds, I tried 
```
esptool.py -p /dev/ttyUSB0 read_mac
```
but with no success:
```
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

Anything more I can do to other than throwing the plug away?

Thx in advance
Georg

tonysprenk, 2020-05-13:
To put the esp chip in flash mode gpio0 should be connected to ground during startup.

waltner, 2020-05-13:
> To put the esp chip in flash mode gpio0 should be connected to ground during startup.

Well, as I was holding the button pressed the two LEDs for `RX` and `TX` on the `CH340G` were red, as soon as I released the button, the lights went off. I thought pressing the buttons is the same as grounding `GPIO0`?

tonysprenk, 2020-05-13:
Only if the module your flashing supports it which the tywe2s doesn’t I believe.

waltner, 2020-05-13:
Thank you very much for the clarification. Grounding `GPIO0` was necessary. I left it soldered until the flash succeeded and removed the wire afterwards. Now everything as usual, going to test it now. :) :+1:

closed 2020-05-13 by waltner

tonysprenk, 2020-05-13:
You’re welcome! Head over to [esp-HomeKit-devices](https://github.com/RavenSystem/esp-homekit-devices) if you want native HomeKit support!

waltner, 2020-05-13:
Thx for the offer. Cool project!

Tbh I have no Apple products at all and am fiddling round with hass.io. :wink:

kueblc, 2020-05-13:
Glad you were able to recover. FYI it was not dead, just running the intermediate firmware, waiting for you to upload your third party firmware.

> ```
> Available options:
>   0) return to stock
>   q) quit; do nothing
> ```
> 
> 
> You can see the available options were limited. ;)

This menu is generated based on the contents of the `files` directory. Did you remove or modify files from this directory?

waltner, 2020-05-13:
Well, atm the directory looks like this and tasmota showed up last time I flashed a bulb:
```
lrwxrwxrwx espurna.bin -> WThermostat_1.06b.bin
-rw-r--r-- espurna.bin_bak
lrwxrwxrwx tasmota.bin -> tasmota.bin_8.2.0
-rw-r--r-- tasmota.bin_8.1.0.2
-rw-rw-r-- tasmota.bin_8.2.0
-rw-rw-r-- tasmota-minimal.bin_8.2.0
-rw-r--r-- upgrade.bin
lrwxrwxrwx WThermostat_1.06b.bin -> ../../Thermostate/BHT-002-GBLW/WThermostatBeca/WThermostat_1.06b.bin
```

So in this intermediate FW mode, I assume the only way to connect is over serial (as I did)?

kueblc, 2020-05-13:
When running the intermediate firmware, it accepts commands via HTTP. This is what the firmware selector uses to start flashing a third party firmware.

You can also send these commands manually, ie
```
curl "http://10.42.42.42/undo"
```
or
```
curl "http://10.42.42.42/flash?url=http://10.42.42.1/files/tasmota.bin"
```

waltner, 2020-05-13:
Nice, good to know. Was a bit tricky with my low quality solder equipment. :grin: