ct-Open-Source/tuya-convert #615: Repeating Resending SmartSocket Packets ... and then end

Kuki77, 2020-05-13:
Dear all,
I'm trying to flash a Sonoff Basic switch, so one of the "basic" gadgets for usage with tasmota, but unfortunately I'm getting a repeating "Resending SmartSocket Packets" message and then the tries are ending.
An other test with a different socket (a Sonoff S26F) ends with the same behaviour, so I suppose it's not about the two devices but about my tuya-convert installation.

Computer:
Acer V5 running a live Debian (freshely created today) with an additional WiFi USB adapter with Linux compatible chipset (Edimax EW-7811Un) and a freshely installed tuya-convert.

terminal output:
`user@debian:~$ cd tuya-convert
user@debian:~/tuya-convert$ sudo ./start_flash.sh 
tuya-convert v2.4.3
Checking for network interface wlx08beac0b0658... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 1282.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 1383.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
sudo: ufw: command not found
======================================================
  Starting AP in a screen.....................
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
...........
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
user@debian:~/tuya-convert$ 
`

Logs:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4624190/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4624191/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4624192/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4624193/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4624194/smarthack-wifi.log)

tonysprenk, 2020-05-13:
Do sonoff devices run tuya firmware?

kueblc, 2020-05-14:
They do not. Sorry @Kuki77 but tuya-convert only supports Tuya devices.

closed 2020-05-14 by kueblc

Kuki77, 2020-05-14:
OK, thanks for clearifying this. My fault, sorry.