ct-Open-Source/tuya-convert #616: Bulb stuck with intermediate firmware

Jaecen, 2020-05-15:
I have a pair of generic wifi light bulbs that look just like [this one](https://www.gearbest.com/smart-bulbs/pp_009302531844.html). No identifying markings on the outside, but I did open one up and confirm it's using an Espressif board.

I put the bulb in flash mode and ran `./start_flash.sh`. After a couple tries, I get a long set of dots after a `Resending SmartConfig packets`, then it just stops and I get the `Device did not appear with the intermediate firmware` message.

When I powercycle the bulb, no `vtrust-recovery` network is visible, however when I have the `vtrust-flash` network up the bulb flickers every ~10 seconds. Tailing `smarthack-wifi.log` shows those flickers are related to immediate connect/disconnect events from the bulb. This makes me believe that it's running the intermediate firmware. I am unable to curl 10.42.42.42 during this time, likely because the bulb doesn't stay connected for more than a fraction of a second.

Any ideas on how to keep the bulb connected to the wifi long enough to upload the new firmware?

```
pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.3
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
sudo: ufw: command not found
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
........................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]

======================================================
Cleaning up...
Closing AP
Exiting...
```

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4632690/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4632691/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4632692/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4632693/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4632694/smarthack-mqtt.log)

Jan003, 2020-05-18:
I have exactly the same problem

tiprya, 2020-05-20:
I am also having the same issue. 

Weirdly, I flashed 6 bulbs that looked identical, and only the final one gave this error.

alejandro5x, 2020-05-24:
Same problem here, from the 4 plugs i bought the last one didn't work, the other 3 i flashed without issues like 3 months ago

AWSb24, 2020-05-24:
Same here. Following!

ranrinc, 2020-07-21:
same issue here.. any idea what happen?

stevenbell, 2020-10-17:
I seem to have run into the same issue.  I'd happily flashed about 30 smartplugs (BN-LINK BNC-60/U133TJ) in a row, when it suddenly started failing with these symptoms.  I don't think it's just that some of these have a newer firmware version or something; I've got way too many plugs to have randomly flashed all the good ones first.  I think the problem started when I plugged in an Ethernet cable, but I don't know.

Reading the reports here, it's always the "last one" that's bad, which makes me think the host configuration is getting messed up somehow.