ct-Open-Source/tuya-convert #623: RM mini IR Bridge - Device did not appear with the intermediate firmware

diogo-madeira, 2020-05-16:
Generic Smart IR, probably RM mini IR Bridge (https://templates.blakadder.com/RM_mini.html). Not known firmware.

I already flashed 6 devices with no problem, including another Smart IR.
With this device I get "Device did not appear with the intermediate firmware" every time I try.

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
..............................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] y
======================================================

I tried many times (at least 6) and I always get the same message.
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4638739/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4638740/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4638741/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4638742/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4638743/smarthack-wifi.log)

dinhchinh82, 2020-05-17:
AFASK RM mini IR use the private wifi chip and it is unable to flash by tuya convert. 
RM mini should be used directly without flashing new firmware with Home Assistant.

diogo-madeira, 2020-05-17:
> AFAIK RM mini IR use the private wifi chip and it is unable to flash by tuya convert.
> RM mini should be used directly without flashing new firmware with Home Assistant.

But in tasmota website (https://templates.blakadder.com/RM_mini.html) say that it was flashed with tuya-convert...

How can I add it to Home Assistant with the original firmware?

CervDotBe, 2020-05-23:
> R
One google way.. https://www.home-assistant.io/integrations/broadlink/

diogo-madeira, 2020-05-23:
> One google way.. https://www.home-assistant.io/integrations/broadlink/

This is only for broadlink smartIR. My is Tuya.

yiannisis, 2020-07-29:
Did you manage to do it?
I'm looking to buy it

https://www.aliexpress.com/item/32961012754.html?aff_platform=portals-tool&sk=_dYc72Dj&aff_trace_key=ad46a8783ef8414cb3077511738cc481-1595436156311-00915-_dYc72Dj&terminal_id=3ac645b4aa5741e4bebe6d5c100f96fc&tmLog=new_Detail&aff_request_id=ad46a8783ef8414cb3077511738cc481-1595436156311-00915-_dYc72Dj

diogo-madeira, 2020-07-29:
> Did you manage to do it?
> I'm looking to buy it
> 

Yes, but not with tuya-convert. I need to open it to flash. I recomend you to buy another model, that can be flashed thru tuya-convert.

plan4sustain, 2020-08-06:
Hello, how about this model https://templates.blakadder.com/auvisio_S06.html

killtux, 2020-10-28:
> > Did you manage to do it?
> > I'm looking to buy it
> 
> Yes, but not with tuya-convert. I need to open it to flash. I recomend you to buy another model, that can be flashed thru tuya-convert.

How did you open it without cracking it? can't find any screws.

just solder the pins to FTDI and flash? is there any tutorial?

diogo-madeira, 2020-10-28:
> How did you open it without cracking it? can't find any screws.
>
> just solder the pins to FTDI and flash? is there any tutorial?

Remove the mat under the device and you will see 4 screws.
No tutorial. Just flash it like any other tuya device.

adrianslabu, 2021-05-20:
Hi, I have one bought from amazon on 22 December 2020, few days ago I found tasmota and home assistant and I managed to flash it via tuya-convert, configured mqtt, tasmota addon detected the device but I'm stuck here...
I recorded and saved all the signals from my remote and they are working in console with IRSend but I don't know how to add them in home assistant to make automatizations... can you please give me some tips? Thank you!