ct-Open-Source/tuya-convert #625: Gosund Sp111 3450W Vtrust-recovery no paring mode

pinkie16, 2020-05-17:
Hi, I tried to flash my Gosund Sp111 3450W it failed and now I have a vtrust-recovery open but cant connect to it or reflash it because it cant bring them back in paring mode. curl http://10.42.42.42/undo doesnt help it says the hole time cant find device when i try to flash.

kueblc, 2020-05-17:
Connect to the `vtrust-recovery` access point with your phone or computer and you can then issue `curl` or use a web browser to access http://10.42.42.42/undo

pinkie16, 2020-05-18:
Tried that in multiple ways connected to both of the vtrust with diffrent phones, tried it with curl also with the webbrowser. 
[image]

[image]

[image]

kueblc, 2020-05-21:
I'm not sure I understand, since the images you posted are of the tuya-convert script.

You need to connect to the `vtrust-recovery` AP, and with that same device, issue `curl http://10.42.42.42/undo` or open that URL with a webbrowser, again from the same device connected to `vtrust-recovery`. This does not involve `start_flash.sh`.

mweinelt, 2020-06-25:
I had the same problem with two SP111 I got delivered mid june. They would flash the intermediate firmware, show up with the vtrust-flash SSID, I could even download the tasmota.bin from it, but as soon as you'd ask it to flash something it would crap itself.

Verified this via serial flash, where both devices I tested just stopped flashing at 15%. I'm sending those back.

mweinelt, 2020-06-29:
Got a new batch of SP111 and retried flashing them. The intermediate firmware gets flashed, the backup is made

```
$ docker-compose exec tuya start
tuya-convert v2.4.4
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlp8s0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen...
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
....../start_flash.sh: line 135:   326 Terminated              ./smartconfig/main.py
............................................................................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0   105k      0  0:00:09  0:00:09 --:--:-- 32687
curl: Saved to filename 'firmware-208393.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 208393
[... 113 lines cut ...]
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 116.243/116.243/116.243/0.000 ms
❯ nmcli
❯ curl 10.42.42.42/undo -v
*   Trying 10.42.42.42:80...
* Connected to 10.42.42.42 (10.42.42.42) port 80 (#0)
> GET /undo HTTP/1.1
> Host: 10.42.42.42
> User-Agent: curl/7.70.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Type: text/plain
< Content-Length: 27
< Connection: close
< Access-Control-Allow-Origin: *
< 
Rebooting into userspace 1
* Closing connection 0
```

After 10 minutes I disconnected and reconnected the SP111, the LED ring stays blue, so it's not on the intermediate firmware anymore. Unfortunately I don't seem to be able to bring it back into pairing mode again. The first press seems to toggle the relay, further presses do nothing. I'm not familiar with the stock firmware, but this feels wrong.

Tried tuya-convert again, but it doesn't find the device. So I guess it will be UART next.

mweinelt, 2020-06-29:
Flashing via UART is stuck again at 15%. What is wrong with these devices? :cry: 

```
 esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 devices/gosund_sp111/208393/.pioenvs/208393/firmware.bin
esptool.py v2.7
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 24:62:ab:20:83:93
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 477488 bytes to 323061...
Writing at 0x00008000... (15 %)
A fatal error occurred: Timed out waiting for packet header
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 115200 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 devices/gosund_sp111/208393/.pioenvs/208393/firmware.bin
esptool.py v2.7
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
```