ct-Open-Source/tuya-convert #626: Bneta Gu10 downlight not flashing

HellfireZA, 2020-05-17:
I'm having trouble flashing the mentioned bulb, I'm also unsure how to attain the logs so could someone please help me with that.

I'm running on Ubuntu 16.04 and have Tuya-convert setup correctly AFAIK, I'm able to connect my phone to the flash network and it says "connected". I then put the bulb into flash mode and it definitely links up to "vtrust-flash" in seconds but doesn't go any further. I can see it on the vtrust-flash network using the Fing app and am able to ping it without a problem. I then changed the IPs in the config from 10.42.42.42 to 10.42.42.22 and it does seem to communicate but connection is refused when backing up and attempting to write firmware. 

Any help would be gladly appreciated.

kueblc, 2020-05-17:
Check for logs in your backup folder

HellfireZA, 2020-05-18:
> Check for logs in your backup folder

I checked my backup folder and a folder is generated it seems every time the bulb does connect to vtrust-flash, all folders are empty though. I've attached screenshots of the backup folder and then what happens the moment I run start_flash.sh

[image]
[image]

HellfireZA, 2020-05-18:
I went and bought another bulb from local vendor and same thing, connection refused but it hooks onto the vtrust-flash network perfectly.

kueblc, 2020-05-18:
Looks like you have an older version of `ss`, and so it is failing to check for occupied ports.

Check those ports manually to make sure they are available, otherwise the software may not be able to start successfully.

kueblc, 2020-05-18:
Also, do not change the IP in the scripts from `10.42.42.42`. It must remain so.

HellfireZA, 2020-05-19:
So an update, I reinstalled Ubuntu to 18.04.4 and everything worked perfectly, I think it was the occupied ports, seemed that Python was using them. Thank you for your help and your work!

kueblc, 2020-05-19:
Excellent! Glad you got it working

closed 2020-05-19 by kueblc

aquarat, 2020-07-29:
Hi

I hope it's okay if I post something to this closed issue: @HellfireZA did you flash Tasmota to your BNeta GU10 device? and if so, what template did you use to control it?

HellfireZA, 2020-07-29:
Hi @aquarat I did to over 30 bulbs now. 
I did experience two different variants of the bulb though.

What you can do is use "generic" and then manually select each pin according to what Tasmota recommends for the Gu10, I found that for one variant of the bulb used: https://templates.blakadder.com/bneta_IO-WIFI60-E27P.html

and the other, the correct gu10: https://templates.blakadder.com/bneta_IO-WIFI-GU10S.html

aquarat, 2020-07-29:
Thanks for the info @HellfireZA, yes I've gone and bought 30 of them too haha, so much flashing.