ct-Open-Source/tuya-convert #628: Gosund SP112 disconnects as soon as flash starts

mhamiltonj, 2020-05-18:
**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

Gosund SP112, current firmware unknown.
Logs: https://gist.github.com/mhamiltonj/fbd613f3f84424669ae9188805fc109e

Using an RPi 2B running Rasbian with a TP-Link TL WN823N usb stick.

AP comes up fine and I can connect and Android device. I can put the SP112 in flash mode, but as soon as the script starts the flashing process, the flashing blue indicator, flickers a few times, then turns off, and the flash fails. Anyone have a similar issue? This device is shown as having been succesfully flashed before in the wiki, does anyone know if it changed?

thelefty, 2020-05-19:
I have this same issue for Treatlife Dimmer switch. It doesn't keep it in pairing mode long enough to flash it. Rather annoying as I know others have flashed these. Any know how to keep it in pairing mode?

kenlasko, 2020-05-20:
I'm seeing the same thing with the Owfeel WiFi Water Control Valve. I'm able to get it into fast-flashing mode. If I don't do anything, the LED flashes for a minute or more. But as soon as I start the flash process thru Tuya-Convert, the LED stops flashing.

gsicilia82, 2020-05-20:
Same issue with Gosund SP112. Tested with Docker flash-method.
SP112 is in flashmode, Smartphone connected to AP, and script is proceeding to "flash".
Something is happend in this time, because the SP112 is not blinking anymore.
But I see only the following screen. I can see this "Smartconfig attempt" few times, but it doesn't proceed to next steps. At the end there is written: ```Device did not appear with the intermediate firmware```
[image]

MyBugYourBug, 2020-05-20:
Same issue wich is mentioned a coupl of times here. I tried to flash a GOSUND SP112 and SP111 without success. After the flash procedure starts the fast flashing led stops. I bought the smart switches by Amazon new yesterday. Is there any chance for a script update in near future?

kenlasko, 2020-05-21:
I saw the exact same thing. As a test to make sure the thing was actually working, I downloaded the SmartLife app and successfully paired my water shutoff valve. The listed firmware was 1.1 and is apparently up-to-date. This is my first foray into flashing ESP, and am willing to help out in any way needed.

I checked the logs while attempting to flash it, and I did see it successfully connecting to the wifi and getting an IP address, but that's as far as it got:
May 20 20:09:02 rpi2 dnsmasq-dhcp[208]: DHCPDISCOVER(wlan0) f4:cf:a2:46:cd:34 
May 20 20:09:02 rpi2 dnsmasq-dhcp[208]: DHCPOFFER(wlan0) 10.42.42.18 f4:cf:a2:46:cd:34 
May 20 20:09:02 rpi2 dnsmasq-dhcp[208]: DHCPDISCOVER(wlan0) f4:cf:a2:46:cd:34 
May 20 20:09:02 rpi2 dnsmasq-dhcp[208]: DHCPOFFER(wlan0) 10.42.42.18 f4:cf:a2:46:cd:34 
May 20 20:09:02 rpi2 dnsmasq-dhcp[208]: DHCPREQUEST(wlan0) 10.42.42.18 f4:cf:a2:46:cd:34 
May 20 20:09:02 rpi2 dnsmasq-dhcp[208]: DHCPACK(wlan0) 10.42.42.18 f4:cf:a2:46:cd:34 ESP_46CD34

kenlasko, 2020-05-21:
Did some further digging. I looked through the scripts and it appears that the scripts are expecting the device to use the IP 10.42.42.42 address. I can see that the device gets 10.42.42.18. As soon as the vtrust-flash AP lights up, it connects via that address. I ran nc -zv 10.42.42.18 8-65534 and the only port that appears open is 6668. As soon as I put the device into fast-flash mode, it no longer responds to pings on that address. I did a scan of the 10.42.42.0 subnet and didn't find anything other than the default gateway at .1 and my phone. Not sure where to go from here.

kueblc, 2020-05-21:
Unfortunately, according to your psk log, this device is running an updated firmware with psk version 2 which is not supported. See #483

closed 2020-05-21 by kueblc

mrmaximas, 2024-04-22:
@kenlasko may be you have original tuya fw backup from this valve? thanks!

kenlasko, 2024-04-22:
Sorry, no I don't. I don't remember exactly what I did after this, but I did eventually get all my Owfeel valves flashed with ESPHome. Been running like a charm for years now.