ct-Open-Source/tuya-convert #630: tuya-convert install fails

IanAdd, 2020-05-19:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

./install_prereq.sh errors with the nessage below, 

Not being a linux geek I have no idea what needs to be done. Help !

THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    pycryptodomex from https://www.piwheels.org/simple/pycryptodomex/pycryptodomex-3.9.7-cp37-cp37m-linux_armv7l.whl#sha256=de048a10052d7a4ec685ab3527fab2c81dc559c0e472be74f16f0171e5c1baa0:
        Expected sha256 de048a10052d7a4ec685ab3527fab2c81dc559c0e472be74f16f0171e5c1baa0
             Got        9ac33ea1ebec1f6a3677fbb930946bbfb6693820fcad39e49c26596e118b301a

IanAdd, 2020-05-19:
Actually, re-running the script has completed without error. Let's see if it works.

kueblc, 2020-05-21:
Let us know if you experience any new issues

closed 2020-05-21 by kueblc