ct-Open-Source/tuya-convert #631: [Feature request] Allow symlinks in .bin files

waltner, 2020-05-19:
Hi,

as reported in #614 I had no option to flash the device after putting the intermediate FW. I know faced the same issue with a different device and looked into the firmware_picker.sh, where there is [a check for file size](https://github.com/ct-Open-Source/tuya-convert/blob/31856ada24a1e951b4da849bff2ed056f544c51b/scripts/firmware_picker.sh#L17).

As I have structured my Firmware through symlinks, none of them shows up:
```
$ l files
total 2,3M
drwxr-xr-x 2 user user 4,0K May 19 09:11 .
drwxr-xr-x 7 user user 4,0K May  8 16:42 ..
lrwxrwxrwx 1 user user   21 Apr  1 13:32 espurna.bin -> WThermostat_1.06b.bin
-rw-r--r-- 1 user user 407K Feb 20 20:53 espurna.bin_bak
lrwxrwxrwx 1 user user   19 May 19 09:11 tasmota.bin -> tasmota.bin_8.1.0.2
-rw-r--r-- 1 user user 453K Feb 20 20:53 tasmota.bin_8.1.0.2
-rw-rw-r-- 1 user user 578K May  8 16:52 tasmota.bin_8.2.0
-rw-rw-r-- 1 user user 375K May  8 16:55 tasmota-minimal.bin_8.2.0
-rw-r--r-- 1 user user 495K Feb 20 20:53 upgrade.bin
lrwxrwxrwx 1 user user  100 Apr  1 13:23 WThermostat_1.06b.bin -> /home/user/software/HomeAutomation/Thermostate/BHT-002-GBLW/WThermostatBeca/WThermostat_1.06b.bin
```

Would it make senso to allow smaller file sizes or allow symlinks (e.g. check if file is a symlink if <1000 bytes)? Or what is the reason behind this check? I understand that flash size might pose restrictions for file size, but one might warn instead of not allowing in the first place.

Best regards
Georg

kueblc, 2020-05-21:
Hi Georg, thanks for the suggestion.

I've added a commit that will include symlinks in the firmware picker. It is available immediately on the `development` branch, and will soon be released on `master`.

The minimum filesize is due to the fact that you must have, at minimum, one sector (4KB) for the bootloader. The maximum filesize is due to the fact we need to divide our 1MB flash space in two, one for the currently running firmware and one for the new firmware by OTA.

closed 2020-05-21 by kueblc

waltner, 2020-05-21:
Thx! Also for the explanation. :+1:

commit 7752f941 (2020-05-21): Follow links, closes #631