ct-Open-Source/tuya-convert #637: Hi I have a lampux smart bulb and I need to change the WiFi how do I do that please

tomwest123, 2020-05-22:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2020-05-22:
This issue tracker is for questions and issues regarding tuya-convert only

closed 2020-05-22 by kueblc

tomwest123, 2020-05-22:
I don’t understand any of that