ct-Open-Source/tuya-convert #638: [Suggestion] NixOS Firewall Troubleshooting

rendeko, 2020-05-22:
Small suggestion, but I noticed for running the Docker container from NixOS, I had to change my NixOS configuration to remove `networking.firewall.enable = false;` and then rebuild. After doing so, my AP was able to distribute an IP address to my phone and I was able to flash. It might be worth noting this along with the lsof command to check for other services such as systemd taking up needed ports.

kueblc, 2020-05-22:
Hi @rendeko thanks for the suggestion. I won't be implementing this myself as I do not use docker or NixOS, but I'm happy to take a look at a PR if you would like to make this change yourself.

Cyber1000, 2021-01-30:
I'm uncertain if this has anything to do with docker itself, since the docker-container uses network: host and privileged: true, it should be similar if executing directly on the host. But I don't know NixOS in depth (we would need to test tuya without docker to know if this is a docker issue), I'll add a comment in the docker section to be aware of possible firewall-problems ...

TheRealGramdalf, 2025-04-05:
I tried this the other day, here's what I found:

- Simply opening the ports listed [here](https://github.com/ct-Open-Source/tuya-convert/blob/master/scripts/setup_checks.sh#L138-L145) (via `nixos-firewall-tool`) didn't seem to work - it would start, find a device, and seemed to flash intermediary firmware, but it wouldn't reconnect to the AP
- It is possible to completely disable the firewall at runtime **without changing your configuration/rebuilding** by running `systemctl stop firewall.service` (NOT `firewalld`)
- I have `systemd-resolved` as part of my configuration, so I also had to stop that for the docker container to start properly
- Additionally `libvirtd` runs a `dnsmasq` server, so I had to kill that via `btop` - simply stopping `libvirtd` didn't work, I believe due to socket activation
  - I would recommend restarting after running `tuya-convert` if you had to disable either `dnsmasq` or `systemd-resolved`, since otherwise DNS resolution seemed to be broken - it would resolve some domains, appearing to be fine, but others didn't. A reboot cleared it all up
  - It should be possible to add a check for a NixOS firewall [here](https://github.com/ct-Open-Source/tuya-convert/blob/master/scripts/setup_checks.sh#L112) and stop `firewall.service` automatically, but that would only work in docker, and I haven't tried running TC directly on the host. Since PRs don't seem to be getting merged (due to PSK rendering TC useless on most newer products) I won't bother opening one unless a contributor can confirm they are available to review/merge