ct-Open-Source/tuya-convert #640: ESP8266 GU10 Bulb - cannot flash

currentsauce, 2020-05-24:
Hi,
I am sorry to add to this common problem, but I've not had any luck trying to fix myself and I'm it's driving me mad. I haven't managed to figure out the problem from the logs.

I have 2 Ajax-Online branded GU10 RGBW bulbs that I am trying to flash. I had a suspicion that maybe they weren't ESP devices, but I've opened one, and it is definitely 8266. The pads are hard to access so i'd really love to get them flashed this way!

The CLI output is below, and logs attached:

> Starting smart config pairing procedure
> Waiting for the device to install the intermediate firmware
> Put device in EZ config mode (blinking fast)
> Sending SSID                  vtrust-flash
> Sending wifiPassword
> Sending token                 ### 00000000
> Sending secret                0101
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> ..............
> Device did not appear with the intermediate firmware
> Check the *.log files in the scripts folder
> Do you want to try flashing another device? [y/N]

Massively grateful for any pointers and help on this,
Thanks!
James

Extra info:
Brand: Ajax Online
Model: AJ-WIFI-GU10-RGBCW
Model Code: SB58

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4674189/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4674190/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4674191/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4674192/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4674193/smarthack-udp.log)

kueblc, 2020-05-24:
Unfortunately your device is not compatible due to #483

closed 2020-05-24 by kueblc

currentsauce, 2020-05-24:
Thank you for saving me from going an endless circle with this! I'll have to do it by wire then. Would it help if I provided the firmware dump of this to help your development or do you already have this one?

kueblc, 2020-05-24:
Firmware samples are always appreciated :) Thank you