ct-Open-Source/tuya-convert #643: Gosund SP1 cant flash it !

xsasx, 2020-05-25:
Hi cant flash my gosund sp1 here are the logs :( 

Freshly installed raspberry pi

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4678107/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4678108/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4678109/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4678110/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4678111/smarthack-udp.log)

kueblc, 2020-05-25:
Did you follow the instructions closely, including running `install_prereq.sh` first?

xsasx, 2020-05-25:
yep:

`pi@octopi:~/tuya/devop/tuya-convert $ ./install_prereq.sh
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Fetched 15.0 kB in 0s (16.3 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.3).
curl is already the newest version (7.52.1-5+deb9u10).
git is already the newest version (1:2.11.0-3+deb9u6).
haveged is already the newest version (1.9.1-5+deb9u1).
iw is already the newest version (4.9-0.1).
libssl-dev is already the newest version (1.1.0l-1~deb9u1).
mosquitto is already the newest version (1.4.10-3+deb9u4).
net-tools is already the newest version (1.60+git20161116.90da8a0-1).
python3-dev is already the newest version (3.5.3-1).
python3-setuptools is already the newest version (33.1.1-1).
python3-wheel is already the newest version (0.29.0-2).
screen is already the newest version (4.5.0-6).
dnsmasq is already the newest version (2.76-5+rpt1+deb9u1).
hostapd is already the newest version (2:2.6-21~bpo9~rpt1).
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 172 not upgraded.
Collecting git+https://github.com/drbild/sslpsk.git
  Cloning https://github.com/drbild/sslpsk.git to /tmp/pip-1nwdm09g-build
Please do not run git as root, your regular user account is enough :)

If you need to run git with root rights for some other application than
what comes preinstalled on this image you can remove this sanity check:

    sudo rm /root/bin/git

You might have to restart your login session after doing that.
Command "git clone -q https://github.com/drbild/sslpsk.git /tmp/pip-1nwdm09g-build" failed with error code 1 in None
pi@octopi:~/tuya/devop/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.3-12-g101bc19
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
`

kueblc, 2020-05-25:
Your logs show missing dependencies, please correct this before trying again

xsasx, 2020-05-25:
which one is missing and how can i install / correct it ? also tried development builed :( im lost

smizles, 2021-07-05:
> which one is missing and how can i install / correct it ? also tried development builed :( im lost

I'm having a similar problem. I see that `./install_prereq.sh` errors out here:
```
Collecting git+https://github.com/drbild/sslpsk.git
  Cloning https://github.com/drbild/sslpsk.git to /tmp/pip-req-build-fya70jk6
Please do not run git as root, your regular user account is enough :)

If you need to run git with root rights for some other application than
what comes preinstalled on this image you can remove this sanity check:

    sudo rm /root/bin/git

You might have to restart your login session after doing that.
Command "git clone -q https://github.com/drbild/sslpsk.git /tmp/pip-req-build-fya70jk6" failed with error code 1 in None
```

I have ran `git clone -q https://github.com/drbild/sslpsk.git /tmp/pip-req-build-y186am_q` manually after running the script without `sudo`, and it executes successfully. However, I am not able to flash my device, and `pi@octopi:~/tuya-convert $ ./start_flash.sh` runs until it errors out with:
```
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/pi/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
........................................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
No ESP82xx based devices connected according to your wifi log.
Here is a list of all the MAC addresses that connected:
1e:d7:3e:4a:72:07

If you see your IoT device in this list, it is not an ESP82xx based device.
Otherwise, another issue may be preventing it from connecting.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] 
```

Does anybody know what the problem is?

smizles, 2021-07-05:
UPDATE: I'm getting a different error from `pi@octopi:~/tuya-convert/scripts/smartconfig $ python3 main.py` after running `sudo pip3 install pycryptodomex` and letting that install. I will try to reflash shortly because the error looks like the error is related to the AP not running. :crossed_fingers: