ct-Open-Source/tuya-convert #644: Odd device behavior requiring 10-15 programming attempts to successfully flash

pookycade, 2020-05-26:
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4679715/smarthack-wifi.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4679716/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4679717/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4679718/smarthack-web.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4679719/smarthack-mqtt.log)
[success-device-info.txt](https://github.com/ct-Open-Source/tuya-convert/files/4679721/success-device-info.txt)

[success-firmware-8f47b2.log](https://github.com/ct-Open-Source/tuya-convert/files/4679723/success-firmware-8f47b2.log)
[success-smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4679725/success-smarthack-mqtt.log)
[success-smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4679726/success-smarthack-psk.log)
[success-smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4679727/success-smarthack-udp.log)
[success-smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4679728/success-smarthack-web.log)
[success-smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4679729/success-smarthack-wifi.log)

### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

Ok, first, I think you all did an awesome job here.  Second, I have something that is just making me pull my hair out.  I am using KMC Wifi Mini 30137.  Flashing it with current build tuya-convert and a wemos D1 for vtrust.  The problem I am having is that this has turned into almost fly fishing here, as in cast the line 30 times and then eventually a fish gets caught.  I can run the start_flash script multiple times and a device will fail, whether I put it into pairing mode, don't put it into pairing mode, plug in other devices in its place.  Sometimes its stragithforward and it will take on the first try. That's rare.  More often than not it takes on the 10th or 15th try.  The log udp seems to indicate it thinks it is NOT a esp device.  BUT, as you can see you eventually latch on and it backs up the firmware and it puts your firmware in its place.  I verify that it shows up as a Tasmota AP and that the menu system is accessible.  So it is an honest to god ESP8266 in there.  But for reasons I can't seem to fathom, more often than not it just doesn't catch and I have to fish again.  Right now I am successfully converting about 3 an hour - thats hard labor.  I have provided logs of failure, and the log of success. You will see even with the log of success it complains that it isn't an ESP8266 device.  The standard response of the device when in pairing mode is that the LED goes out after about 5 seconds, and thats it.  I try to repair again and then give up, unplug and go to the next one.  Eventually it catches.  Sometimes not all the way, as in we don't get to the firmware backup, but you can see the ........ of the attempted flash slow down to know that its talking with the device. You can see the device tries to disconnect multiple times.  I did worry about weak connection or noise, but note that 1) doesn't matter if  I do this well away from the router, the Wemos is led solid throughout. Doesn't matter whether I restart the program, pull the plug on the Rasp Pi.    This is obviously very confusing, but I post this in the hope that looking at the firmware and looking at the logs might tell you something more general beyond my specific situation.  It seems if you bang your head against a wall and just hammer the device it will eventually give way to OTA reprogramming, but the wheres, whys, hows are all a mystery to me.  I cannot identify a single set of circumstances that allow successful flashing here.  Just glad it works, but boy is it eating up hours in the process.  Up to device 58 by now after all weekend at it.

kueblc, 2020-05-26:
Your logs seem to show a few ESP82xx devices, and a lot of non-ESP82xx devices, all connected at the same time. This might explain the hit-or-miss behavior you're observing.

I can tell you with relative certainty that the tests for ESP82xx devices has been accurate. We look for strings that are not present in the Tuya ESP SDK.

The firmware you posted for `gwId 74466530c44f338f47b2` shows in the UDP and web logs without the non-ESP warning, as expected.

pookycade, 2020-05-26:
Thanks for the reply.  A few clarifications: 1) the devices are only connected one at a time. I do not however wait for for a full attempted flashing session to finish its 10 attempted cycles before switching to a different plug.  I found that if it doesn’t get it in that round, waiting for 10 full repeated cycles of start_flash never works.  2) I have slowly worked my way thru this group of plugs even though it seems to say that they are not esp8266.  That is to say initially we started with 45 that wouldn’t program at all and are now down to 20 or so.  4) You clearly know more than I do, so I’ll trust you that if they say they are non ESP8266 then they are indeed non ESP8266, but annecdotally some that seem to have been reported non 8266 seem to have eventually programmed.  Just raising the question of whether the SDK got updated possibly here or firmware messaging changed so that that detection is perhaps not entirely accurate.  I can’t prove it one way or another for now, just a gut impression, but I’ll try to pay more attention to the log output for each one and see if I get it to eventually take even if it says it’s non esp8266