ct-Open-Source/tuya-convert #645: Can't flash LoraTap SC400W switches || can't pair device with pi

TomDuyf, 2020-05-26:
**Are you reporting a successful flash?**
No

**Do you have a question about configuring your newly flashed device?**
No

**Did you have issues getting a device to flash?**
Yes. I recently bought three LoraTap SC400W Curtain switches which I'm trying to flash using tuya converter. I followed DigiblurDIY's guide step by step and were able to reproduce everything until the point at which the device is supposed to pair with the pi. 

In short, I start the tuya converter, connect my smartphone to the pi AP, I put the switch in fast pairing mode and press enter on the pi to start the process. The LED indicates communication between the devices because the LED went from fast blinking to solid and and then off while the pi keeps searching for the device. Ending in the following:
`SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
`

I've checked the logs and googled the issue, but haven't found a solution. I nearly concluded that the device must not use an ESP28xx, until I found a thread here [https://github.com/ct-Open-Source/tuya-convert/issues/423](https://github.com/ct-Open-Source/tuya-convert/issues/423) where a user stumbled across an issue for the same device and succeeded in flashing them. The only problem is that I can't follow what he did in order to make it work.

Hope to find a solution to my problem here. Any ideas? 

Attached my logs.
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4684584/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4684585/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4684586/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4684587/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4684588/smarthack-mqtt.log)

**Please remember to close your issue when the problem has been addressed!**

hoenes, 2020-06-02:
Hello TomDuyf,
I was running into the same issue my self some days ago. As I've done serial flashing on other devices before, I ended up soldering and flashing the LoraTap the hard way. It worked (but was a mess, I can tell you if you're interested). Now it seems, that one of the relays are broken. As I only own a single LoraTap maybe you can confirm something for me?

There only 5 electrical connections between the high voltage part (relays and low voltage power supply) and the low voltage part (ESP826x/TYWE2S and button control) I figguerd out (from left to right) Pin 1 and 2 are interconnected and low voltage GND, Pin 3 triggers the up relay, Pin 4 triggers the down relay (in my case NOT, because I think it is broken), Pin 5 is +V3.3. To confirm, what I can't, shortening the Pins 4 and 5 should trigger the down relay as shortening the Pins 3 and 5 triggers the up relay. thx
hoenes

hoenes, 2020-06-02:
[image]
PIN OUT

Fjuxx, 2020-06-02:
I think this issue relates directly to #483 

I have the exact same problem. I think there is no other way than to desolder the 5 pins, and then flash the TYWE2S directly.
If anyone finds a good manual/tutorial that would be great!

kueblc, 2020-06-02:
According to your logs @TomDuyf,
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```

This is a reliable test. Even if other devices of the same model have used ESP82xx in the past, manufacturers have been switching to pin-out compatible chips such as the RTL8710. If you have reason to doubt this test, please provide photos of the chip and we can identify it.

Without an ESP82xx, you will not be able to run ESP firmware, so flashing by serial will not help.

TomDuyf, 2020-06-02:
> Hello TomDuyf,
> I was running into the same issue my self some days ago. As I've done serial flashing on other devices before, I ended up soldering and flashing the LoraTap the hard way. It worked [...]

@hoenes Sadly, my electrotechnical skills are really limited, I even have trouble understanding your situation/request. So I'm afraid I can't help you with your request.

@kueblc When i google the numbers on the chip I'm fairly sure the switch uses a TYWE2S. Isn't that suppose to use a ESP8285?

kueblc, 2020-06-02:
Checking model numbers does not guarantee the chipset. Manufacturers have been violating FCC by not issuing new FCC IDs when they change chipsets.

> Even if other devices of the same model have used ESP82xx in the past, manufacturers have been switching to pin-out compatible chips such as the RTL8710

Your logs indicate that it is not an ESP82xx, so while it is true that the TYWE2S has used an ESP in the past, it seems either your model does not use a TYWE2S or that the TYWE2S in your device does not use an ESP.

TomDuyf, 2020-06-03:
Fair enough, I will just install them over the tuya cloud. Thanks for the replies.

closed 2020-06-03 by TomDuyf

Foxdelmar, 2020-07-28:
I also couldnt flash it via WLAN.

I also bought the following product: https://www.amazon.de/LoraTap-Zeitschaltuhr-Rolladenzeitschaltuhr-Rolladenmotor-Kompatibel/dp/B07L8TF88C

I would like to add my screen output and a picture of the device. To make it easier for others to compare:
[image]
[image]

If anyone finds a way to put another firmware onto the device please share it with me/us!

I have two things i want to get rid of:
1. The device uses a yellow LED that is always on and is shining bright in my sleeping room. It might be turned off using the Tuya cloud what I want to avoid
2. The device starts the roller shutter but it always stops after 10 seconds. So my blinds stop in the middle of the windows. I guess it must be calibrated and this only seems to work with the app that must be connected to the Tuya cloud. The 10sec must be set to at least 25sec.