ct-Open-Source/tuya-convert #646: tuya convert teckin sp22

borux, 2020-05-28:
Hi. I have a problem flashing tasmota on a smart socket. The error is as follows:
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4698014/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4698015/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4698016/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4698017/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4698018/smarthack-mqtt.log)

Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
........Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
......
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder

borux, 2020-05-28:
Can anybody help me? thank you very much

kueblc, 2020-05-28:
From your web log
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```

tuya-convert, Tasmota, Espurna, ESPHome, etc only support ESP82xx devices.

closed 2020-05-28 by kueblc