ct-Open-Source/tuya-convert #649: new style Merkury Plug - MI-WW218 - Not able to flash. What is this chip? Pics/Logs

ctrl50, 2020-05-29:
Hey got some new merkury plugs that wont flash.

Opened one up and This is what I have:

I have a CP2102 will I be able to flash it? Not even sure what chip this is. Google isnt finding much on the p/n and  model# 219979
[image]
[image]
[image]

logs:
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4704745/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4704746/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4704747/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4704748/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4704749/smarthack-mqtt.log)

Anyone have any Info on this? Should i hook it up to teh cp2102 and see what happens or is this a non-flashable chip?

never flashed any esp stuff with serial.. so any pointers? tasmotizer?

ctrl50, 2020-05-29:
by the way.. they dont want you opening this thing.. it took a while with a hacksaw.. its salvagable.. but its not pretty.

kueblc, 2020-05-29:
Unfortunately this is not an ESP82xx as evidenced by the MAC address. You will not be able to install ESP82xx firmware (Tasmota, Espurna, etc) on this device. Sorry to be the bearer of bad news!

closed 2020-05-29 by kueblc

ctrl50, 2020-05-29:
alright.. well that sucks.. RIP cheap merkury plugs.. Guessing they will update the lights to this chip next.

Thanks.

Didgeridrew, 2020-07-14:
> alright.. well that sucks.. RIP cheap Merkury plugs.. Guessing they will update the lights to this chip next.

They have updated the bulbs now... at least mine are.

#692

ctrl50, 2020-10-10:
> > alright.. well that sucks.. RIP cheap Merkury plugs.. Guessing they will update the lights to this chip next.
> 
> They have updated the bulbs now... at least mine are.
> 
> #692

Yup. I got some with the WB2L that wont flash.

[image]