ct-Open-Source/tuya-convert #650: Stuck flashing Teckin SP20

brickedkludge, 2020-05-30:
**DEVICE:** Teckin SP20
**FIRMWARE VERSION:** Didn't think to check before I started flashing- can't connect to Smart Life app anymore to check.

I'm using a Linux Mint 19.3 live distribution on an HP laptop to run Tuya Convert. At first I was using the built-in laptop wifi adapter but could never see the vtrust-flash SSID and start_flash.sh never seemed to do anything.

You can see some representative shell output from one of these attempts in the attached **NoSSID_ShellOutput.txt** file.

After searching for ideas, I tried using an old Netgear WN111V2 USB wifi adapter I have laying around. Then I could see the vtrust-flash SSID and could connect to it with my Android phone.

The first time I ran start_flash.sh while I could see the vtrust-flash SSID I got the following shell output before :
```
<snip>
SmartConfig complete.
Resending SmartConfig Packets
...../start_flash.sh: line 129: 16286 Terminated              ./smartconfig/main.py
...................................................................^C....^C.^C.^C.^C.^C.^C.^C.^C.^C.^C.^C.^C.^C.................Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
<snip>
```
You can see the full shell output in **FirstTry_ShellOutput.txt**

After this, the smart plug didn't behave normally anymore. When I plug it in, I only get a brief red/blue flash. Holding the button while plugging it into power didn't do anything. Pressing or holding the button after plugging it in didn't do anything. Pressing the button no longer results in the mechanical relay toggling. I figured I bricked it.

After some more searching and reading, it sounded like it was worth plugging it in and trying again.

So I tried again and now it seems like it connects to the smart plug and starts trying to fetch a firmware backup before a similar looking error. Then it gets stuck in what looks like a firmware backup progress display until it times out or I exit. Here's the shell output:

```
IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
./start_flash.sh: line 129: 19644 Terminated              ./smartconfig/main.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 1024k    0  4380    0     0     51      0  5:42:40  0:01:25  5:41:15     0^C======================================================
Cleaning up...
Closing AP
Exiting...
```

You can see full shell output in **SecondTry_ShellOutput.txt**

I'm also attaching the logs after the last failures.

Any ideas?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4704943/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4704944/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4704945/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4704946/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4704947/smarthack-wifi.log)

[FirstTry_ShellOutput.txt](https://github.com/ct-Open-Source/tuya-convert/files/4704940/FirstTry_ShellOutput.txt)
[NoSSID_ShellOutput.txt](https://github.com/ct-Open-Source/tuya-convert/files/4704941/NoSSID_ShellOutput.txt)
[SecondTry_ShellOutput.txt](https://github.com/ct-Open-Source/tuya-convert/files/4704942/SecondTry_ShellOutput.txt)

kueblc, 2020-05-30:
Unfortunately this device is not an ESP82xx as evidenced by the MAC address in your wifi log, and therefor will not work with tuya-convert or ESP firmware like Tasmota.

brickedkludge, 2020-05-30:
A couple (hopefully short) questions if you don't mind:

**First question:**
I saw Teckin SP20 in the [compatibility list](https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices-(HTTP-firmware)), so I thought it was compatible. 

Should it not be in that list or do some SP20's have the ESP82xx and some do not?

**Second Question:**
I have some other smart plugs I was hoping to convert. Two are model W-US003 and one looks identical but with zero labelling. I see W-US003 listed [here](https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices-(HTTP-firmware)) and [here](https://templates.blakadder.com/gdtech_W-US003.html) as compatible.

Is there a way I can tell if they really are compatible before I try to convert them and possibly brick them?

Thanks!

kueblc, 2020-05-30:
> Should it not be in that list or do some SP20's have the ESP82xx and some do not?

Yes, they have switch chips so while older models are running on the ESP82xx, newer ones are not. See #269 

> Is there a way I can tell if they really are compatible before I try to convert them and possibly brick them?

Chances of bricking are low, tuya-convert will simply fail on these devices. You can check the MAC address of the device to see if the manufacturer is "Espressif Inc".

> My SP20 still seems to partially connect to Tuya Convert and presumably still has the stock firmware since Tuya Convert gets stuck trying to fetch a firmware backup.

This shouldn't happen if the device is not an ESP82xx, but your logs indicate otherwise. Can you check to see if there are logs in your `backup` folder?

> Is it possible to "unbrick" it by letting it go back to using the stock firmware that is still on the smart plug?

If it is indeed stuck on the intermediate firmware, you can go back to stock firmware by issuing `curl "http://10.42.42.42/undo` or visiting that address in a web browser while the device is connected.

brickedkludge, 2020-05-30:
Thank you for the reply!

> Chances of bricking are low, tuya-convert will simply fail on these devices. You can check the MAC address of the device to see if the manufacturer is "Espressif Inc".

When I look at smarthack-wifi.log, I see these MAC addresses:
```
Using interface wlx0026f28ce028 with hwaddr 00:26:f2:8c:e0:28 and ssid "vtrust-flash"
wlx0026f28ce028: AP-STA-CONNECTED 68:c6:3a:e6:52:ef
wlx0026f28ce028: AP-STA-CONNECTED 2e:95:ab:a7:58:b6
```
When I check 68:c6:3a:e6:52:ef on http://maclookup.app (I assume that is the smart plug), it shows the manufacturer as "Espressif Inc". **If that is correct, then how can you tell it is not an ESP82xx?**

Also, what device is 2e-95-ab-a7-58-b6? It doesn't seem to be my laptop or phone or the smart plug. **What else could it be?**

> This shouldn't happen if the device is not an ESP82xx, but your logs indicate otherwise. Can you check to see if there are logs in your backup folder?

The only file in the backups folder is **firmware-e652ef.bin** with size 4.1kb. Github won't let me attach a file with that extension, but if you want it let me know how to get it to you.

I do have another set of log files from my "FirstTry" described above where I think the smart plug originally got messed up. I am attaching them in case they are helpful.

> If it is indeed stuck on the intermediate firmware, you can go back to stock firmware by issuing curl "http://10.42.42.42/undo" or visiting that address in a web browser while the device is connected.

While running start_flash.sh I was able to ping 10.42.42.42, but when I tried curl "http://10.42.42.42/undo" the command hung and trying to hit it in the web browser just failed to load that URL.

**Does that mean it is bricked?**

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4705050/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4705051/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4705052/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4705053/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4705054/smarthack-wifi.log)

kueblc, 2020-05-30:
> When I [lookup 68:c6:3a:e6:52:ef](https://maclookup.app/search?mac=68%3Ac6%3A3a%3Ae6%3A52%3Aef) it shows the manufacturer as "Espressif Inc". I assume that is the MAC address of the smart plug. If that is correct, then how can you tell it is not an ESP82xx?**

My mistake, I must have missed `68:c6:3a` looking at this late last night. You are correct that it is an Espressif OUI, I will reopen this issue.

> Also, what device is 2e-95-ab-a7-58-b6? It doesn't seem to be my laptop or phone or the smart plug. **What else could it be?**

No idea. It could be a device that randomizes their MAC address for anti-tracking, like a phone.

> The only file in the backups folder is **firmware-e652ef.bin** with size 4.1kb. Github won't let me attach a file with that extension, but if you want it let me know how to get it to you.

You can attach `.bin` files inside a zip archive. Though in this case the backup is probably not useful.

> While running start_flash.sh I was able to ping 10.42.42.42, but when I tried curl "http://10.42.42.42/undo" the command hung and trying to hit it in the web browser just failed to load that URL.

Well it is good news that you can ping it. After you issue the command, does the behavior of the device change? Does it allow you to enter pairing mode after that?

brickedkludge, 2020-05-30:
Here's a zip of the firmware-e652ef.bin in case it happens to be helpful:
[firmware-e652ef.zip](https://github.com/ct-Open-Source/tuya-convert/files/4706178/firmware-e652ef.zip)

Is the fact that the smart plug MAC address starts with 68:c6:3a a guarantee that it has an ESP8266 or could it still have a different chip that is not compatible?

> Well it is good news that you can ping it. After you issue the command, does the behavior of the device change? Does it allow you to enter pairing mode after that?

I don't see any change in the behavior of the device. Still just a single blue/red flash when I plug it in and it is unresponsive to pressing or holding the button.

It occurred to me that I bought two of these identical smart plugs at the same time and both have been connected to the Smart Life app since I got them. The other one that still works reports the firmware as:
No updated available
Wi-Fi Module: 1.0.3
MCU Module: 1.0.3
I assume the now-broken one had the same firmware versions before I tried Tuya Convert on it.

kueblc, 2020-05-30:
> Is the fact that the smart plug MAC address starts with 68:c6:3a a guarantee that it has an ESP8266

Yes. Your backup also confirms this, both because a backup would not be possible otherwise and also because the backup contains the ESP bootloader.

When you reached the backup step, and it failed, did you press `Y` to continue anyway?

You could retry `start_flash` (at this point, connecting another device to `vtrust-flash` is not necessary) and then skip the backup.

brickedkludge, 2020-05-30:
No, when the backup failed previously I just stopped then.

I tried `start_flash` again now and continued after the backup timed-out, but still no luck. I'm attaching my shell output and all the logs from the scripts directory:

[ShellOutput.txt](https://github.com/ct-Open-Source/tuya-convert/files/4706254/ShellOutput.txt)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4706255/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4706256/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4706257/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4706258/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4706259/smarthack-wifi.log)

I don't know if it's useful, but here's also a zip of the most recent backup directory:
[backups_most_recent.zip](https://github.com/ct-Open-Source/tuya-convert/files/4706262/backups_most_recent.zip)

Scope666, 2020-06-17:
I'm in the exact same boat, same exact symptoms, the quick flash when first plugging in, no AP being offered, can't ping it, can't hold button to enter pairing mode,  Dead in the water.