ct-Open-Source/tuya-convert #651: Issues OTA flashing a new Gosund SP112

sfudeus, 2020-05-30:
First: Thanks for this project - I was successful flashing the Gosund SP111 before and have fun using them with Tasmota. 

In addition I now have a fresh set of Gosund SP112 (those with USB), and while I know of users having been successful with them, I seem to have PSK issues. 
I tried to find another fitting issue to just add my part, but I didn't, so I opened another issue, hope that is ok. 

I end up with the message "Device did not appear with the intermediate firmware". 
The culprit seems to be in the `smarthack-psk.log`: 
```
new client on port 443 from 10.42.42.27:31802
ID: 0242416f68626d643661473931494652319edaf954622c145b3009636e2a675d49f2a1eb1c105799fd09f8d5315bc9718e
PSK: 97e520ab470c4d3cc545dad1af5992a65e4bf459d2f50fea8ad849f265e1b5a7
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.27:399
ID: 0242416f68626d643661473931494652319edaf954622c145b3009636e2a675d49f2a1eb1c105799fd09f8d5315bc9718e
PSK: 97e520ab470c4d3cc545dad1af5992a65e4bf459d2f50fea8ad849f265e1b5a7
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.27:64520
ID: 0242416f68626d643661473f3f3f3f3f3f9edaf954622c145b3009636e2a675d49f2a1eb1c105799fd09f8d5315bc9718e
Prefix: b'BAohbmd6aG??????'
PSK: 0c6177f9cea23fc9e6988fdbf2d77af3b80fdc38a0606b9f31f560c31ff65ddb
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.27:19216
ID: 0242416f68626d643661473931494652319edaf954622c145b3009636e2a675d49f2a1eb1c105799fd09f8d5315bc9718e
PSK: 97e520ab470c4d3cc545dad1af5992a65e4bf459d2f50fea8ad849f265e1b5a7
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.27:13050
ID: 0242416f68626d643661473931494652319edaf954622c145b3009636e2a675d49f2a1eb1c105799fd09f8d5315bc9718e
PSK: 97e520ab470c4d3cc545dad1af5992a65e4bf459d2f50fea8ad849f265e1b5a7
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.27:6875
ID: 0242416f68626d643661473931494652319edaf954622c145b3009636e2a675d49f2a1eb1c105799fd09f8d5315bc9718e
PSK: 97e520ab470c4d3cc545dad1af5992a65e4bf459d2f50fea8ad849f265e1b5a7
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.27:11037
```

If you need more infos or want something tried out, I'll gladly comply. 
In case that question comes up, I was using an older RasPI 2 with recent raspbian using OpenSSL 1.1.1d  10 Sep 2019.

kueblc, 2020-05-30:
Hi Stephan, thanks for the data point. This is #483

closed 2020-05-30 by kueblc

sfudeus, 2020-05-30:
Dang, could have matched that, I did see it. Sorry for the noise.