ct-Open-Source/tuya-convert #653: Cannot Flash

StupidNamez, 2020-05-30:
Hi,

I am trying to flash a wifi plug socket, model: LSPA7

My phone is connected vtrust-flash and the socket is slow flashing but nothing happens:

`
IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] 

`
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4705659/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4705660/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4705662/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4705663/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4705664/smarthack-wifi.log)

Any help would be greatly appreciated.
Thanks

kueblc, 2020-05-30:
This is not an ESP82xx as evidenced by the MAC addresses in your wifi log.

closed 2020-05-30 by kueblc

StupidNamez, 2020-05-30:
Can you explain what you mean by that please?

kueblc, 2020-05-30:
tuya-convert and ESP firmware such as Tasmota, Espurna, and ESPHome, only work on the ESP82xx microcontroller, so it is not currently possible to flash this device.