ct-Open-Source/tuya-convert #655: Firmware flash not working on LSC smart plugs that have been used in the LSC app

djmvt, 2020-05-30:
After successful flashing of 4 smart plugs (2 from LSC and 2 generic Chinese with powermonitor) last january I tried to flash 2 LSC smart plugs today that have been in use with the official LSC app. Unfortunately this doesn't seem to work anymore, probably because the exploit is no longer working?

I can start start_flash.sh, using a donor to connect to vtrust-flash and after pressing enter tuya convert reports "sending SSID, sending WiFi password, Sending token, Sending secret".

Then a bunch of dots and finishing with "device did not appear with the intermediate firmware".

In the logs I can see it simply cannot find the smartplug. (Smartplug is in fast blinking mode).

When I try to connect it to the LSC app it is working.

Trying this on my raspberry pi 3b+ with latest raspian full installation, also tried it with a raspian lite clean install but still not working. 

Tried both the master branch and the development branch.

Any ideas?

djmvt, 2020-05-30:
Looks like this is the same as issue #483 
Is it still possible to flash directly on the chip? Haven't found a decent manual for that, everywhere I search I'm getting tuya ota instructions.

kueblc, 2020-05-31:
Please share your logs for assistance

kueblc, 2020-06-20:
Closing until we have more information

closed 2020-06-20 by kueblc

djmvt, 2020-06-20:
It is indeed the same issue as #483, same goes for a curtain switch I received today. Although it is in the list with compatible devices it cannot be flashed anymore.