ct-Open-Source/tuya-convert #658: Unable to flash Feit 6" RGBW Recessed Downlight

danecreekphotography, 2020-06-03:
**Are you reporting a successful flash?**
Sadly no.

**Do you have a question about configuring your newly flashed device?**
No.

**Did you have issues getting a device to flash?**

Name: Feit Electric 6in. RGBW Recessed Downlight
Model number: LEDR6/RGBW/AG
Template: https://templates.blakadder.com/feit_electric-LEDR6_RGBW_AG.html

**Relevant logs**

Looking through all the log files this appears to be the only thing of interest:

`new client on port 443 from 10.42.42.33:22751
ID: 0242416f68626d64366147393149465231e946592a954157d890d1cd412cdb58dcd756a33ec67380fb8ba3c86b90286c4a
PSK: 5a6a8ca5628e389bf220928865946bc683f1d2f5267d5e3ec69d210dcc58816e
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)`

**Additional comments**

I am able to get the light into fast flash mode by turning it on/off 5 times, it flashes multiple times per second. It will stay flashing until I fire up tuya-convert, then it will stop and go solid. Tuya-convert just keeps retrying over and over again.

I've used this same tuya-convert/raspberrypi/nodeMCU combo to flash four plugs successfully so it does appear to be something specific to this light.

danecreekphotography, 2020-06-03:
Looks like the same issue as https://github.com/ct-Open-Source/tuya-convert/issues/483?

rkallal, 2020-06-04:
I have the same Feit Electric 6in. RGBW recessed downlight and have the same issue. I agree this is  #483.

danecreekphotography, 2020-06-04:
I got a second one from Amazon today, hopeful that it would work, and it's also got the same issue.

I can't figure out how to crack the case to do a direct serial flash either. Sad.

rkallal, 2020-06-05:
I busted one of mine apart and broke it trying to solder to the GPIO0, TX, RX pins (I suck at small connection point soldering). It is a well made product but was not that hard to open. The round LED front cover pops out and it only has a 4x 1-inch ridges and a little glue holding the front cover on. Once the front cover is removed, there are only 3 screws holding the back plastic cover on to the metal body. Once you remove those 3 screws the back plastic cover comes apart and the whole thing easily falls apart. 

Using the the back side as a reference, i.e. the side that goes into the ceiling can, if you measure about 1 inch from the spring bracket following edge of the spring bracket, that is where the front cover has those 4 uniform ridges and glue that I talked about above. I completely broke my back cover off before understanding how it went together (don't attempt anything with the back cover until your have the front cover removed). You should be able to pop off the front cover using a very thin 1/2 inch knife pushing in to the edge between the front metal and front plastic cover somewhere need one of the 4 ridges (not visible until you take it apart).

danecreekphotography, 2020-06-05:
Ahhh cool, thanks. Too bad I already returned both to Amazon :D

fyrmangeoff, 2020-06-29:
can you assist with where you soldered to for the connections to flash  I have included pictures for reference?
[image]
[image]

rkallal, 2020-06-29:
I have not succeeded in soldering and flashing the downlight (I suck at small pin soldering). There is a specification on the pinning for the tuya module that contains the ESP8266.

mannysanch, 2020-07-17:
@fyrmangeoff Any pointers on how to get "glass" off these ligths? Want to try to figure out the pins, but am having difficulties opening it up. This was intended to be a sacrificial device, but would still like to practice removing it as cleanly as possible for when I try it on the others I have. Was going to try heating it up, but I'm thinking the cover will warp.

eohlde, 2020-10-04:
I can report that the4 LEDR6/RGBW/AG(C) variant (purchased 09/2020) is much harder to disassemble. The front 'plastic' is now 360 degree adhered. I can't think of a decent way to get this off without destroying it.