ct-Open-Source/tuya-convert #659: Curtain switch

swil85, 2020-06-04:
hello, 
i'm trying to flash a [curtain switch ](https://fr.aliexpress.com/item/32963064005.html?spm=a2g0s.9042311.0.0.3b936c37QYasvP).

and loop on smarthack, here are the log :
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4730475/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4730476/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4730477/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4730478/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4730479/smarthack-wifi.log)

what m'i doing wrong ?
Thank's for help.

simulot, 2020-06-20:
Hi @swil85 
I have bought 2 batches of those devices. 
I have flashed the first batch without problem. Carton boxes have the date  23 FEB 2020
I'm not able to flash those from the second batch, have the date 24 APR 2020.

I have opened a new device. I can't see any difference between both. the tuya reference is still same TYWE3S. So I guess the hardware haven't change, but the firmware now blocks tuya converter.    
Such a shame

kueblc, 2020-06-20:
Yes, unfortunately blocked by #483. Hopefully we discover a workaround!

closed 2020-06-20 by kueblc