ct-Open-Source/tuya-convert #660: would like to return to stock tuya firmware

plowking86, 2020-06-07:
hi guys i would like to return to the stock firmware that tuya-convert backed up but there is little info
on how to go about it 

ive been running tasmota on a light switch was great now want to go back 

what i have done is hard wired it to the serial adapter flashed the backup tuya-convert has made
in the wondows command line 

esptool.exe --baud 115200 write_flash -fs 1MB -fm dout 0x000000 c:\firmware-5ebc3b.bin

but when i turn it on a wifi ap comes on saying Vtrust backup???

any help would be grate cheers

kueblc, 2020-06-07:
You're almost there! Just connect to the AP and visit http://10.42.42.42/undo

plowking86, 2020-06-07:
thanks kueblc that did the trick

just for others if you want to go back to stock his what i did hrs of trial and error

to load stock firmware i did this in windows 10 command make sure you locate where python espinstaller.exe is installed mine was 
C:\Program Files\Python38\Scripts\   your machine might be diffrent so in my command Line looked like this 

C:\Program Files\Python38\Scripts> esptool.exe --baud 115200 erase_flash hit enter
esptool.py v2.8
Found 1 serial ports
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ***********this will be your Mac address
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.1s
Hard resetting via RTS pin...
then while your still in command

C:\Program Files\Python38\Scripts> esptool.exe --baud 115200 write_flash -fs 1MB -fm qio 0x000000 **Location of the bin witch for me was** C:\firmware.bin then hit enter

esptool.py v2.8
Found 1 serial ports
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: **********your MAC
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 1048576 bytes to 510813...
Wrote 1048576 bytes (510813 compressed) at 0x00000000 in 45.4 seconds (effective 185.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

this will load the stock firmware from either from the tuya-convert backups you had stored or if you did a back up with through the esptool.exe 

Now verify the flash is the same 

C:\Program Files\Python38\Scripts> esptool.exe --baud 115200 verify_flash -fs 1MB -fm qio 0x000000 C:\firmware.bin
esptool.py v2.8
Found 1 serial ports
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ********your MAC
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Verifying 0x100000 (1048576) bytes @ 0x00000000 in flash against C:\firmware.bin...
-- verify OK (digest matched)
Hard resetting via RTS pin...

now power up the device you will see a wifi AP connect to it
then visit http://10.42.42.42/undo as Kueblc has said in the above comments 
the device will then disconnect from the AP when that happens power off device and power back on
and your done back to stock

thanks again for help Kueblc for help cheers

closed 2020-06-07 by plowking86