ct-Open-Source/tuya-convert #661: LSC Smart Led Strip

ManiacTwister, 2020-06-09:
Hello,

yesterday i bought 3 LSC Smart Led Strips from Action. Unfortunately the device model is not the same as anymore and the flash over OTA doesn't work - i already found several issues about new firmware revisions so i think this is expected. The only interesing thing i saw is the following message after starting flash OTA, the strip stops blinking and prints the following message several times on uart:

`tcp connect failed m1-US.iot-dns.com:4433 errno=104`

No i tried to flash by uart, unfortunately i can't get esptool to find my device:

`esptool.py --port /dev/ttyS0 flash_id`
`/dev/ttyS0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header`

I have connected gpio0 to ground and CH_EN to vcc. The following is visible on uart:
[console.txt](https://github.com/ct-Open-Source/tuya-convert/files/4750426/console.txt)
[image]

Is it possible that the new revision of the led strip doesn't use and esp? But why does it stop blinking when starting the OTA update then?

The new device model number is 970751

kueblc, 2020-06-09:
> Is it possible that the new revision of the led strip doesn't use and esp?

The [WB3S](https://docs.tuya.com/en/iot/device-development/module/wifibt-dual-mode-module/wb3s-module-datasheet) is not an ESP82xx.

> But why does it stop blinking when starting the OTA update then?

It stops blinking when it pairs with the network, which is the same process on many Tuya devices, not just their ESP based devices. The registration and upgrade process will fail if the device is not an ESP, which is good because otherwise you might brick devices with the wrong firmware.

closed 2020-06-09 by kueblc

sistein, 2020-07-29:
Since this is the only thread I can find when looking for a alternative firmware. Any ideas where to find one?

ManiacTwister, 2020-07-30:
@sistein 
> Since this is the only thread I can find when looking for a alternative firmware. Any ideas where to find one?

If you don't mind soldering: The WB3S is pin compatible with the ESP8266, so you can replace the chip of the led controller board. There is just one change needed to keep IR functionality because the new revision of the board uses another gpio pin for the ir led:

```
code/espurna/config/hardware.h
#elif defined(LSC_SMART_LED_LIGHT_STRIP)
-    #define IR_RX_PIN           0
+    #define IR_RX_PIN           3

```

Nosstrom, 2020-09-17:
Hello. I got he same problem with tuya and wb3s. I already got a tywe3s chip, if i solder it, do you think that might work?