ct-Open-Source/tuya-convert #666: Problem with neo coolcam 16A nas-wr01w power plug

tty74it, 2020-06-13:
I've buyed 2 of this power plug.
One is flashed without problem.
With the second one i can't flash it, and it reboot many time when tuya-convert is activated

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4774328/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4774329/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4774330/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4774331/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4774332/smarthack-wifi.log)

Thank's

kueblc, 2020-06-13:
Did you buy these two plugs at the same time?

Would you share the logs and firmware backup from the other device?

tty74it, 2020-06-13:
Yes same time
[device-info.txt](https://github.com/ct-Open-Source/tuya-convert/files/4774634/device-info.txt)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4774635/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4774636/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4774637/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4774638/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4774639/smarthack-wifi.log)

kueblc, 2020-06-13:
Can you post the backup as well?

tty74it, 2020-06-13:
Yes in the zip

Thank's

kueblc, 2020-06-13:
I do not see the registration call in your web log. Have you tried putting the device back in pairing mode and starting from scratch?

Is this a clean tuya-convert install or have you added or changed any files?

tty74it, 2020-06-13:
Now i've tried win a cleaninstall and any files are changed (only config wlan interface name WLAN=wlx00156d846131)

But is the same, i see the device reboot many time and also the relay status ( click sound on reboot i  think)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4774697/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4774698/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4774699/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4774700/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4774701/smarthack-wifi.log)

kueblc, 2020-06-13:
I see that these two devices are running different firmware, and that the first device was converted in April. It may be that you have gotten a firmware update from Tuya since then, blocking tuya-convert.

tty74it, 2020-06-13:
Ok thank's , i'm waiting a new version tuya-convert, because i see is not to simple solder wire in this device.
At the moment i use the first device converted

Many thank's

nogueirasa, 2020-07-23:
Hello, any news? I'm with the same problem, new firmware from Tuya.

stefxx, 2020-08-28:
I just ordered 5 of those plug. I was only able to tuya-convert 2 of them successfully. I will have to wait for my serial adapter before I can attempt serial flashing the other three.

Too bad. OTA flashing is sooooo much easier. Really hope a future version will fix this.

worldwidefelix, 2020-08-28:
> I just ordered 5 of those plugs. I was only able to tuya-convert 2 of them successfully. I will have to wait for my serial adapter before I can attempt serial flashing the other three.
> 
> Too bad. OTA flashing is sooooo much easier. Really hope a future version will fix this.

Hi. Do you get any information how to open the plug with latest one? It seems like there are some differences between previous model. The new one is really hard to get into under the PCB which is pad for flashing.

stefxx, 2020-08-28:
Nope, haven't tried yet.

worldwidefelix, 2020-08-28:
> Nope, haven't tried yet.

I see. Hope to share how to work out this matter altogether. Thx

stefxx, 2020-08-28:
Just opened the top cover. I was hoping it would be simple but I can't find any screws to get it out of it's housing.

Might need to sacrifice one to see how we can open them neatly.

KoalaBear84, 2020-08-28:
I do not have experience with it, but found this last week:
https://github.com/xoseperez/espurna/wiki/Hardware-NEO-COOLCAM-NAS-WR01W

stefxx, 2020-08-28:
I've seen that as well. But it is not the same version (10A vs 16A version). It can't locate the two screws to get the boards out of the housing.

stefxx, 2020-08-28:
Turns out there is only the one screw. Hammer carefully on the pins and the whole module will come out. Watch out for the earth connector (at the screw point), thats all. Unfortunately, I am afraid I already ruined it, but that's ok.

Here are some pictures:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

You can clearly see my damage on the second picture, the black part is broken. Anyway, I was able to find GND, 3V3, RXD and TXD (TXD is accessible but the label is hard to see below the relay. I can't find GPI00, I am afraid we need to take it apart to get to it.

worldwidefelix, 2020-08-28:
> Turns out there is only the one screw. Hammer carefully on the pins and the whole module will come out. Watch out for the earth connector (at the screw point), thats all. Unfortunately, I am afraid [...]

Thank you for sharing your experience.
Mine was already got damaged ,either while I tried to open.
I have no idea after opening top cover at that time.

At the first, I was looking for another 2 screws.
Now, I know how to disassembly finally.

sanderlv, 2020-10-01:
any luck or any schema available to do the flashing?

djfanatix, 2020-10-11:
I have the same issue, bought 3 plugs, 2 worked fine, but the 3rd switches of the pairing mode and the relay flickers when Tuya convert tries to connect.

sanderlv, 2020-10-24:
Just asking... any update on this device? I have successfully flashed 16 other devices now running esphome :-)... Wanting to use these 2 as well...

stefxx, 2020-10-24:
Nope. We have to wait until tuya-convert works again on newer devices (if ever). I tried to open one and in order to flash them serial I most certainly destroy them in the process, so I decided to wait. Again, it might be a looooong wait.

sanderlv, 2020-10-24:
Argh... I hate to wait (long), I tend to forget then :-)

bobybobybob, 2020-11-10:
it is less difficult to solder the Flash component. Then solder on the circuit board of an ESP01, put Tasmota on it, unsolder it and put it back into the china socket.

stefxx, 2020-11-11:
Hi @bobybobybob ! Do you have some instructions/pictures on how you flashed the Neo Coolcam?

bobybobybob, 2020-11-15:
I have two of these adapters. One broke when I took it apart. From this broken one I unsoldered the flash module. From one ESP01 the flash component was desoldered and the other one was soldered. With Tasmotizer the Tasmota is loaded. Desoldered the component again. 

With the other adapter with a side cutter I removed the flash component and soldered the rest. Soldered the Tasmota Flash module into the adapter and finished. Set the Tasmota Template etc. and everything is great.

bobybobybob, 2020-11-15:
https://s12.directupload.net/images/201115/uf5v5zdj.jpg

stefxx, 2020-11-15:
Thanks @bobybobybob , also for the great pictures. Not sure if I am brave enough to do that, but then again, I broke one already and the others are currently unused because I can't flash them... And if not, I am sure it will help others!

Gunner251192, 2020-11-20:
Hi, when running Tuya convert it seems to regocnize the plug, because it stops blinking fast or you hear the relay clicking.
I got 10! of them ( NAS-WR01W 16A plugs bought 3 weeks ago) and really want them to flash so i can use them local in my automations. It would be really great to get this fixed, to help i uploaded my logs when running Tuya Convert, if there is someone who can point me in the right direction it would be really great. They are pretty useless now.

Thanks in advance.
(https://github.com/ct-Open-Source/tuya-convert/issues/705)

Shamshala, 2020-12-02:
> @stefxx ... I can't find GPI00, I am afraid we need to take it apart to get to it.

If you haven't already found out, you don't need to look anywhere for GPIO0 actually, it's connected to the button. I'm kinda impressed that you been able to hammer it out. I took the harder way to desolder the top PCB inside the socket in my 6 plugs. :rofl: 

I tried TuyaConvert two times, it bricked my sockets two times but i already knew how took them apart so it wasn't a big deal.

Shamshala, 2020-12-02:
> @stefxx ... I can't find GPI00, I am afraid we need to take it apart to get to it.

If you haven't already found out, you don't need to look anywhere for GPIO0 actually, it's connected to the button. I'm kinda impressed that you were able to hammer it out. I took the harder way to desolder the top PCB inside the socket in my 6 plugs. :rofl: 

I tried TuyaConvert two times, it bricked my sockets everytime but i already knew how took them apart so it wasn't a big deal and flashed them directly.

clausmuus, 2020-12-07:
Thank you all for your documentation. So it was possible for me to flash my devices successful. For the second one I need only a view minutes :)

gabest11, 2020-12-09:
I have one with ID: 02... that do not work with tuya-convert. Desoldered the flash chip and saved the original fw if anyone needs it. Can I just load it with files/tasmota.bin even though it's less then 0.5 MB and the original is 1 MB?

clau-bucur, 2020-12-21:
> 
> 
> Thank you all for your documentation. So it was possible for me to flash my devices successful. For the second one I need only a view minutes :)

I've just flashed tasmota-minimal.bin using Tasmotizer.
It creates the wifi, I can connect to it, I get ping replies from 192.168.4.1 but I can't connect to it to configure it. No ports are opened (even after a couple of hard resets).

Can you describe your flashing procedure? What tools and what firmware have you used ?
Thanks!

clau-bucur, 2020-12-21:
Nevermind, it works now :)
I gave it another shot and flashed by selecting image "Release / tasmota.bin" and checking "Erase before flashing".

finalstate, 2020-12-23:
Don't use a hammer to open the plug :-)
After unscrewing and taking the 'female' side off:
- either use a vise: put the plug into it, and slowly close it. Two thirds of the plug pins length pushed in should be enough to get it out by hand
- even simpler, if you have a drill press, use it to push the innards out. Be careful not to damage the bit holder.

alnavasa, 2021-01-01:
I have flashed It soldering the pins but I might have damaged the step-down voltage converter, Or I don't know what, but, I have two flashed ones with erratic behavior:
1: I plug it, it works ok, I un-plug it, and plug it back, nothing happens, doesn't work. un-plug it, wait 10 min, plug it back, it now does work.

2: Plug it, works for 15 seconds, then it powers of. 

In My case is a "Avatto 16A Power Monitoring Plug" from aliexpress. but its the same probably

I am wondering if I broke the step-down converter, or what the hell happened...

[image]
[image]

ragzol, 2021-01-20:
I have flashed! It soldering the pins tx, rx, gnd, 3.3V pcb and connect serial usb adapter. 
flashing: 
Connect pin to usb adapter: tx->rx, rx->tx, 3.3v -> 3.3v, gnd-> gnd
GI00 -> gnd
switch push or connect two switch pins

tasmota pyflasher run

esptool.py v2.8
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 50:02:91:d7:4a:95
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Compressed 602192 bytes to 430094...
Wrote 602192 bytes (430094 compressed) at 0x00000000 in 46.9 seconds (effective 102.8 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.

[image]
[image]
[image]

stefxx, 2021-01-20:
Flashed, no soldering. It takes a few attempts to get the pins just right ;)

[image]
[image]
[image]

pesor, 2021-01-26:
@stefxx 

> Flashed, no soldering. It takes a few attempts to get the pins just right ;)
> 

Where did you get that smart clamp from?

stefxx, 2021-01-26:
@pesor https://nl.aliexpress.com/item/4001016093009.html?spm=a2g0s.9042311.0.0.7b1c4c4dqvC5IW

pesor, 2021-01-26:
@stefxx 

Thanks, it is ordered. Got 6 Avatto 16 Amp, all with new firmware, so this is a really nice solution without soldering, and I gladly wait for delivery.

When I look at your picture, what did you do with the GI00, it is not clear from your photo.

stefxx, 2021-01-26:
Just press the button during startup.

alnavasa, 2021-01-26:
> @stefxx
> 
> Thanks, it is ordered. Got 6 Avatto 16 Amp, all with new firmware, so this is a really nice solution without soldering, and I gladly wait for delivery.
> 
> When I look at your picture, what did you do with the GI00, it is not clear from your photo.

Let us know how it goes, 
I don't know What I did to my Avatto, but they booted and after a minute or so died.
Tasmota was working on them, but after the minute they where gone...
I got tired of trying to flash and reflash them

alnavasa, 2021-01-26:
> @pesor https://nl.aliexpress.com/item/4001016093009.html?spm=a2g0s.9042311.0.0.7b1c4c4dqvC5IW

what size did you order?

stefxx, 2021-01-26:
I have it for a while, I think I got the largest one. But if it is just for flashing the Neo Coolcam, it's larger than necessary.

pesor, 2021-01-26:
12 pins double row, 24 pins in total, that is the same size as @stefxx

Sendt fra min iPhone

Fettkeewl, 2021-01-27:
Hio

I ordered 2 PCS of theese. And through some very informative posts here I've flashed tasmota on them. 
They state that it should be power monitoring on theese ones but I'm guessing that's not true? 
Does anyone know? 

SEK 32.37  36%OFF | Smart Plug WiFi Socket 3680W 16A Power Energy Monitoring Timer Switch EU Outlet Voice Control By Alexa Google
https://a.aliexpress.com/_mqxbfQV

KoalaBear84, 2021-01-27:
That is true, they have Power Monitoring.

Fettkeewl, 2021-01-27:
> That is true, they have Power Monitoring. I would like to order some more, but I don't dare as I don't want to open them and flash them through hardware.. :(

Alright, anyone know what pin and module the power monitoring is on? I've reassembled mine.. Would rather not open again. 

BTW it's fairly easy to open them up, hardest part is soldering the small pads. I'd recommend making a template with pogo pins.

gabest11, 2021-01-27:
Just got a few more of these plugs and the only v1 tuya firmware I had no longer works. I flashed two plugs and none of them turn on. Don't know if I did something wrong. It was quicker for me to swap the spi flash chip with hot air than fiddling with the small solder points.

I use this template for esphome. https://textuploader.com/18jqq

alnavasa, 2021-01-27:
> Just got a few more of these plugs and the only v1 tuya firmware I had no longer works. I flashed two plugs and none of them turn on. Don't know if I did something wrong. It was quicker for me to [...]

do they behave like mines? 

> I have flashed It soldering the pins but I might have damaged the step-down voltage converter, Or I don't know what, but, I have two flashed ones with erratic behavior:
> 1: I plug it, it works ok, I [...]

gabest11, 2021-01-27:
It does nothing at all, but I was thinking about it and I may have used the BlitzWolf SHP6 firmware backup. That has a different GPIO configuration. Going to try to reflashing it later. The inside of the plugs were identical to the old ones, except they put a black sleeve on the earth where the screw goes and also a black sticker next to the relay to hide the soldering points.

pesor, 2021-01-28:
I use a Vice to gently press the legs out. 

Before you do it, you have to remove the screw, and the top cover.

[image]

Shamshala, 2021-01-28:
@pesor And what about pressing it back? Do the insides hold tightly in the case without wiggle?

pesor, 2021-01-28:
Do it like this, use a plug, not another Awatto, but a big plug.

[image]

yvesdm3000, 2021-02-13:
Beware the 16A version has a screw now at the front:

[image]

Yves

stefxx, 2021-02-13:
The 16A always had the screw at the front. At least all the ones I got over the last 12 months.

pesor, 2021-02-13:
Next time I need smart plugs, I will buy these, already flashed with Tasmota

ATHOM preflashed TASMOTA Smart Plug Works With Home Assitant Electric Consumption Monitoring 16A
https://a.aliexpress.com/_mN4M2wj

garynl, 2021-02-13:
This is great news! I prefer the coolcam's because of their compact size,
but this is a good development

[21 more replies]